alienatedsec/yi-hack-v5 #313: Followed steps in a Yi 1080p Home camera 2023.10 and don't works.

aicastell, 2023-11-11:
**Describe the bug**
I have a Yi 1080p Home camera, running stock firmware 9.0.19.12_202102241808. Followed the steps to reflash with your custom firmware, and it didn't work. 

**To Reproduce**
Followed described steps. Used a 16GB SD card. Created a single FAT32 partition (fdisk option b). Formated the partition with FAT32 using gparted. Downloaded the firmware yi_home_1080p_0.3.8.tgz. Extracted and copy the content in the root directory. Connected the SD to the camera and powered the device. I can see the MAC of the camera in the ARP cache of my computer, and have the IP. The ping to the IP of the camera works fine. However, when I try to open http://<IP>:8080, it doesn't work. I did a nmap -Pn <IP> and there is any exposed service. 

**Expected behaviour**
The camera shoud be running the new firmware, but it seems not working fine. 

**Set Up Details (please complete the following information):**
 - Firmware Version: 0.3.8
 - Device: YI 1080p home camera
 - Date: 2023/10

** Extra **
I can provide any detail you need. Even I can break the safety sticker, I bought this camera just to play with it. Many thanks!

aicastell, 2023-11-11:
Attached is a picture of the electronic:

[image]

aicastell, 2023-11-11:
Already fixed! Just needed to use a different repository. This worked for me. (ID 

https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues

Just used this to choose my correct firmware:
    - DID: RFUSY440...
    - Stock firmware: 9.0.19.12_202102241808.

Many thanks!

closed 2023-11-11 by aicastell

poseidogithub, 2024-04-15:
> Already fixed! Just needed to use a different repository. This worked for me:
> 
> https://github.com/roleoroleo/yi-hack-Allwinner-v2/
> 
> Just used this to choose my correct firmware: - DID: RFUSY440... - Stock firmware: 9.0.19.12_202102241808.
> 
> Many thanks!

I have the same firmware but I can't make the bloddy thing to flash.

I've downloaded Yi 1080p Home | BFUS - IFUS - RFUS | 9.0.19* | y21ga | -
unconmpressed it to a formated 8gb fat32 sd card, plug it on the camera, and... I only see orange flashing light and waiting to connect annoying sound. any hint?