alienatedsec/yi-hack-v5 #321: vlc cannot open the rtps stream of an yi home camera (27us)

allegfede, 2023-12-07:
**Describe the bug**
vlc cannot open the rtps stream of an yi home camera (27us)

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to main camera web page: http://192.168.1.61:8080/index.html
2. copy the url from the High res stream: in may case rtsp://192.168.1.61/ch0_0.h264
3. open vlc and paste the copied url in the "open net stream" 
4. press play or what you got in your language
5. error window pops up vlc unable to open MRL 'rtsp://192.168.1.61/ch0_0.h264'

**Expected behaviour**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Set Up Details (please complete the following information):**
 - Firmware Version: [e.g. 0.3.8]
 - Device: [e.g., YI HOME Camera]
 - Model: [e.g., 27US]

**Additional context**
Add any other context about the problem here. If any other systems are involved, please complete the following:
 - Name of the system [e.g., Home Assistant]
 - Version [e.g., 2022-12-xx]

**Technical Details**
Please add any technical details (e.g., log files)

alienatedsec, 2023-12-08:
@allegfede Not enough information - please update your post with missing information and share the outcome of the `ps` command from the console

allegfede, 2023-12-10:
Shure. how can i do that? there is a debug section in the cam web interface?

closed 2024-01-17 by github-actions[bot]