alienatedsec/yi-hack-v5 #323: Change Onvif daemon

lebestnoob, 2023-12-24:
I've noticed that this project uses roleoroleo's [onvif_srvd](https://github.com/roleoroleo/onvif_srvd) application. It works fine, but it really only works to send video across. The same project also looks like it has been abandoned by roleo in favor of his much newer [onvif_simple_server](https://github.com/roleoroleo/onvif_simple_server). He has also been using for his yi-hacks. From what I understand, this server runs as a cgi so the web server can share the same port as the Onvif server and not be separate like how it currently is, unless there is another reason to have it separate. The server also appears to support Onvif events like motion detection, which would be helpful in situations where the camera is added to an NVR. Could it be possible to implement it?

alienatedsec, 2023-12-25:
@lebestnoob I see no issues about implementing it. I will look into it after the new year.

alienatedsec, 2024-02-03:
@lebestnoob that's in progress https://github.com/alienatedsec/yi-hack-v5/commit/e258274218c85f064b925c9e481c911d7fb8451e

alienatedsec, 2024-02-04:
Please follow and test if needed https://github.com/alienatedsec/yi-hack-v5/releases/tag/0.4.1e

lebestnoob, 2024-02-05:
Thanks. I'll post my results sometime this week

lebestnoob, 2024-02-06:
It appears that the "Enable/disable time OSD" toggle no longer works. I looked through the code and found the the logic responsible for enabling this function was deleted in the `system.sh file` . The previous firmware used to have the code below as part of the `get_config DISABLE_CLOUD` function:
``` 
if [[ $(get_config TIME_OSD) == "yes" ]] ; then
            echo -ne '\x01\x00\x00\x00' | dd of=/tmp/mmap.info bs=1 seek=0 count=4 conv=notrunc
fi
```
but this is missing. I have amended this code locally and it seemed to have reenabled the timestamp.

alienatedsec, 2024-02-10:
@lebestnoob something went wrong when I did the merge and resolve conflicts - I hope this will be back to what it was before. However, I still cannot reproduce the `TIME_OSD` as it does not work for me. Which camera do you have?

lebestnoob, 2024-02-10:
I have Yi Cloud Dome. It seemed to work fine for

alienatedsec, 2024-02-11:
Does it work when you upgrade to the latest pre-release? You will need to do a manual upgrade

alienatedsec, 2024-02-14:
@lebestnoob Please can you test the ONVIF with `0.4.1i`?

lebestnoob, 2024-02-16:
I've updated the camera to the latest firmware and ONVIF has been working fine so far. I've added it to a Hikvision NVR and it is recording it there correctly. It is also viewable on Tinycam (android app) when using the generic ONVIF profile.

alienatedsec, 2024-02-19:
This is now implemented, so please raise another issue if needed.

closed 2024-02-19 by alienatedsec