alienatedsec/yi-hack-v5 #326: OTA FW upgrade shows no value for "Available:"

reef-actor, 2024-01-20:
**Describe the bug**
When running on 0.4.0 I noticed both the "Latest" and "Pre-Release" available values were blank and trying to upgrade resulted in "Firmware download failed". I recently reflashed 0.38 and the issue was still there. I have now manually upgraded to 0.4.1d (by copying the tgz) and still have the issue.

 - Firmware Version: 0.4.0, 0.4.1d
 - Device: Outdoor 1080
 - Model: 9CUS

The camera has unrestricted internet access.

Are there any log files that might reveal any issues?

g1za, 2024-01-22:
I experienced the same problem and I also recently re-flashed to v.0.3.8. When I tried to update to the latest pre-release from the interface the problem appeared again so I'm updating manually too.

alienatedsec, 2024-01-22:
> Are there any log files that might reveal any issues?

Unfortunately, no log files to check - however, you could run the command from the console and see what will be reported back.

Release check
```
/usr/bin/wget -O - https://api.github.com/repos/alienatedsec/yi-hack-v5/releases/latest 2>&1 | grep '"tag_name":' | sed -E 's/.*"([^"]+)".*/\1/'
```

Pre-release check
```
/usr/bin/wget -O - https://api.github.com/repos/alienatedsec/yi-hack-v5/releases 2>&1 | grep -B 4 '"prerelease": true' | awk -F '"' '{print $4; exit}'
```

g1za, 2024-01-22:
For me the output of the release command is 0.4.0, while for the pre-release is 0.4.1d

alienatedsec, 2024-01-22:
It could be the browser then - does not allow to execution of javascript

g1za, 2024-01-22:
I remember I used to launch updates from the interface and JS seems to be enabled.
BTW... it it was a JS issue, should not we experience the same in saving settings, deleting recordings and so on?

[image]

alienatedsec, 2024-01-22:
> I remember I used to launch updates from the interface and JS seems to be enabled. BTW... if it was a JS issue, should not we experience the same in saving settings, deleting recordings and so on?
> 
> ![immagine](https://private-user-images.githubusercontent.com/88257868/298651602-21eec39e-85fd-4fcc-aebf-f7b9e732d14e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU5NDE3MzgsIm5iZiI6MTcwNTk0MTQzOCwicGF0aCI6Ii84ODI1Nzg2OC8yOTg2NTE2MDItMjFlZWMzOWUtODVmZC00ZmNjLWFlYmYtZjdiOWU3MzJkMTRlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTIyVDE2MzcxOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBjMzIzY2VkZTJiMjI3NWRiYTk4NmEwMzI3ZjBlYzZkMzRkYWFjNDA1YTcwYTllN2M3OWE1NDAzNDdiYjEwNjQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.b1VcxtNi0WlqJpPslv0LUTlzVWrypdV7Fe4hwC-K18E)
> 
> EDIT: I also reloaded some previously saved settings file from the interface

Firefox? Maybe try another browser?

MichlW, 2024-01-22:
The same happens to me on all cameras. Even in ssh console its the same empty answer:
```
/home/yi-hack-v5 # /usr/bin/wget -O - https://api.github.com/repos/alienatedsec/yi-hack-v5/releases/latest 2>&1 | grep '"tag_name":' | sed -E 's/.*"([^"]+)".*/\1/'
/home/yi-hack-v5 #
```

reef-actor, 2024-01-22:
> There is an TLS error:

This is the same issue I have

g1za, 2024-01-23:
Apologies I need to amend my previous comment.
With SSH I also get empty answers:
`/home/yi-hack-v5 # /usr/bin/wget -O - https://api.github.com/repos/alienatedsec/
yi-hack-v5/releases/latest 2>&1 | grep '"tag_name":' | sed -E 's/.*"([^"]+)".*/\
1/'
/home/yi-hack-v5 # /usr/bin/wget -O - https://api.github.com/repos/alienatedsec/
yi-hack-v5/releases 2>&1 | grep -B 4 '"prerelease": true' | awk -F '"' '{print $
4; exit}'
/home/yi-hack-v5 # 
`

alienatedsec, 2024-01-23:
I started experiencing the same issue recently - something broke and need to find what :)

roleoroleo, 2024-01-24:
busybox must be updated to 1.36.1
This will fix the problem.

alienatedsec, 2024-01-24:
Thanks @roleoroleo 

I started yesterday and will need to rework the whole baseline to prevent this and some other issues.

Rushmed, 2024-01-29:
How to update busybox?

alienatedsec, 2024-01-29:
@Rushmed you have to wait for another release

reef-actor, 2024-02-09:
> Please follow and test if needed https://github.com/alienatedsec/yi-hack-v5/releases/tag/0.4.1e

[image]

Brilliant work!

closed 2024-02-09 by reef-actor

alienatedsec, 2024-02-11:
Please beware that a new baseline is not final, so upgrades via WebUI are prevented until finalised with the latest release.

I just need two more issues fixing to release the latest version.

muava12, 2024-04-22:
i still in ver 2.3.8, and baseline ver 0.3.8. can i manually put 0.4.0 somewhere using FTP and update the firmware.. since i far away from my cctv right now