alienatedsec/yi-hack-v5 #329: Manual Updlaod button for firmware

fraserp, 2024-01-30:
**Is your feature request related to a problem? Please describe.**
I wanted to check if a PR to fix the MQTT warnings in HA worked, but to do so I'd need to put a manually edited firmware version on the cameras. But my camera is mounted high up so I'd have to physically turn it on, climb up, remove the SD card.... etc. So to make life easier could we have a manual FW upload button. Would also be useful in cases like this bug: https://github.com/alienatedsec/yi-hack-v5/issues/326 

**Describe the solution you'd like**
An 'upload firmware' button on the  'Service' screen

**Describe alternatives you've considered**
I tried doing this via SSH but got frustrated with trying to get wget to see the place I'd put the file. so gave up. (Kept getting a 550 error and gave up working out why)

alienatedsec, 2024-02-03:
@fraserp a bit outdated (legacy v4) but still available to upload `rootfs` and `home` files - it will work if FAT32 formatted card is inserted
```
http://192.168.x.x:8080/?page=update
```

I am planning to modify and include uploading of an upgrade package.

fraserp, 2024-02-04:
I was actually looking to upload a modified 0.4.1 tgz to include an as yet unmerged PR that fixes the MQTT errors in Home Assistant, so I could help by testing if it works... so I don't thank thats going the help me in this instance, right? But thanks anyway!

alienatedsec, 2024-02-25:
There is no chance to have a button that will upgrade the baseline at the same time, so it's only possible for non-baseline-related upgrades. As the latest `0.4.1` is now released, I will work on this moving forward.

reef-actor, 2024-02-29:
@alienatedsec have you experimented to see if it is possible to support two partitions on a card, one FAT32 and one exFAT so that flashing is possible without wiping the card?

alienatedsec, 2024-02-29:
@reef-actor I was thinking about it recently but not followed the idea or even trying. I am not sure how the embedded system sees multi-partitions or even mounts them. I'll have a look.