alienatedsec/yi-hack-v5 #334: [0.4.1.e] - Yi Home 27US - Unable to Flash

Minims, 2024-02-04:
**Describe the bug**

I have follow the how to and format a 16GB in FAT32. I have tried the windows solution, but also with my own tools. I have tried more than 10 times and everytime the led stay yellow. I have wait about 30min.

there is no new files on the SD.

On reboot without the sd card or with the 0.4.1d sd, it’ok.

**To Reproduce**
Steps to reproduce the behaviour:
Follow the 0.4.1e guide

**Expected behaviour**
be able to flash

alienatedsec, 2024-02-04:
@roleoroleo is your model also not flashing with the release files?

roleoroleo, 2024-02-05:
Not tested yet.
I'll let you know as soon as possible.

alienatedsec, 2024-02-08:
I will need to make the home image smaller - it's too big to accept flashing

roleoroleo, 2024-02-08:
I compared your image with mine and there is a big difference in size: about 1,5 MB.
The main problem is the library /home/app/libcrypto.so.1.1
I don't know why but in my image is not present.

alienatedsec, 2024-02-08:
I'll have a look in the evening.

alienatedsec, 2024-02-10:
> The main problem is the library /home/app/libcrypto.so.1.1

@roleoroleo did you mean `/home/lib/libcrypto.so.1.1`? It must be there as needed by the latest WPA unless you are not compiling it and it's not being replaced.

[image]

alienatedsec, 2024-02-10:
[New release - 0.4.1f](https://github.com/alienatedsec/yi-hack-v5/releases/tag/0.4.1f)

Minims, 2024-02-10:
Thanks, i was able to flash the camera. It seems working excepted the known bug of the pre-release.

closed 2024-02-10 by alienatedsec

alienatedsec, 2024-02-11:
@Minims is 0.4.1g also flashing and not connecting on your 27US?

Minims, 2024-02-11:
@alienatedsec 0.4.1g is flashing and wifi is ok on my side with wpa2

alienatedsec, 2024-02-11:
@Minims 
When you type wpa_supplicant in the console, does it report v2.10?

Minims, 2024-02-11:
@alienatedsec Yes
```
/home/yi-hack-v5 # uptime
 14:48:36 up 44 min,  load average: 4.97, 4.79, 4.39

/home/yi-hack-v5 # cat /tmp/wpa_supplicant.conf
ctrl_interface=/var/run/wpa_supplicant
ap_scan=1
network={
ssid="********"
scan_ssid=1
proto=WPA RSN
key_mgmt=WPA-PSK
pairwise=CCMP TKIP
group=CCMP TKIP
psk="********"
}

/home/yi-hack-v5 # wpa_supplicant
Successfully initialized wpa_supplicant
wpa_supplicant v2.10
Copyright (c) 2003-2022, Jouni Malinen <j@w1.fi> and contributors
```