alienatedsec/yi-hack-v5 #335: Beeping on 0.4.0, Yi Outdoor Cam

databoy2k, 2024-02-07:
**Describe the bug**
I have a Yi Outdoor 1080p camera, flashed with 0.4.0. When I updated from 0.3.8, I started to get a weird issue that whenever anything walks in front of the camera, it beeps loudly.

I have disabled the ONVIF stream and only use RTSP. I do not have audio enabled. Recording and cloud are disabled, and the only advanced config enabled is SSH, HTTPD, and Swap File. Under "Settings" I have the camera "on", I do not save video when motion is detected, detection sensitivity is low, and status and IR leds are both on. Rotate is off. 

I only access the camera through Frigate, which does local object recognition. I cannot find anywhere that the camera should be doing its own detection.

**To Reproduce**
Honestly, not sure; if the bug is in place, just walk in front of the camera.

**Expected behaviour**
Camera should not beep.

**Screenshots**
N/A

**Set Up Details (please complete the following information):**
 - Firmware Version: 0.4.0
 - Device: Yi Outdoor 1808p
 - Model: 9CUS

**Additional context**
Add any other context about the problem here. If any other systems are involved, please complete the following:
 - Name of the system: Frigate
 - Version: 0.13.1-34FB1C2

**Technical Details**
(let me know if you need any logs)

alienatedsec, 2024-02-07:
Please try the latest pre-release

databoy2k, 2024-02-07:
Is it possible to do that without having to pull the microSD and reformat using yi app? These are two steps that are pure headache for my setup.

alienatedsec, 2024-02-08:
@databoy2k As mentioned in the release notes, don't try if not convenient.

Minims, 2024-02-08:
Duplicate with https://github.com/alienatedsec/yi-hack-v5/issues/298 ?

databoy2k, 2024-02-08:
> Duplicate with #298 ?

Yep/ @alienatedsec these can be merged. I searched "beep" but didn't search "sound" or "alarm" ;)

alienatedsec, 2024-02-14:
duplicate with #298

alienatedsec, 2024-02-14:
Duplicate with #298

closed 2024-02-14 by alienatedsec