alienatedsec/yi-hack-v5 #336: [0.4.1e] No timestamp watermark on stream

Minims, 2024-02-08:
**Describe the bug**

My Yi Outdoor has no timestamp water by default. It was working in 0.4.1d. 

**To Reproduce**
Steps to reproduce the behaviour:
1. Enable OSD watermark
2. Reboot
3. Stream rtsp and check there no watermark 

**Expected behaviour**
having a timestamp watermark on stream

**Set Up Details (please complete the following information):**
 - Firmware Version: 0.4.1e
 - Device: YI OUTDOOR 1080P
 - Model: 9CUS
 
[image]

lebestnoob, 2024-02-09:
I encountered the same issue. I wrote my solution here. It involves manually editing the `system.sh` file in the sdcard in the `yi-hack-v5/script` folder, so I only advise doing it if you feel comfortable editing it.  You can take out the sdcard or edit it through ftp by accessing `/tmp/sd`. A restart will be required.
https://github.com/alienatedsec/yi-hack-v5/issues/323#issuecomment-1928794469

Find the DISABLE_CLOUD sections (should begin line 165) and replace it with this:
```sh
if [[ $(get_config DISABLE_CLOUD) == "no" ]] ; then
    (
        cd /home/app
        killall dispatch
        LD_PRELOAD=/home/yi-hack-v5/lib/ipc_multiplex.so ./dispatch &
        sleep 3
        if [[ $(get_config TIME_OSD) == "yes" ]] ; then
            echo -ne '\x01\x00\x00\x00' | dd of=/tmp/mmap.info bs=1 seek=0 count=4 conv=notrunc
        fi
        LD_LIBRARY_PATH="/home/yi-hack-v5/lib:/lib:/home/lib:/home/app/locallib:/home/hisiko/hisilib" ./rmm &
        sleep 8
        ./mp4record &
        ./cloud &
        ./p2p_tnp &
        if [[ $(cat /home/app/.camver) != "yi_dome" ]] ; then
            ./oss &
        fi
        ./watch_process &
    )
fi
if [[ $(get_config DISABLE_CLOUD) == "yes" ]] ; then
    (
        cd /home/app
        killall dispatch
        LD_PRELOAD=/home/yi-hack-v5/lib/ipc_multiplex.so ./dispatch &
        sleep 3
        if [[ $(get_config TIME_OSD) == "yes" ]] ; then
            echo -ne '\x01\x00\x00\x00' | dd of=/tmp/mmap.info bs=1 seek=0 count=4 conv=notrunc
        fi
        LD_LIBRARY_PATH="/home/yi-hack-v5/lib:/lib:/home/lib:/home/app/locallib:/home/hisiko/hisilib" ./rmm &
	sleep 8
        if [[ $(get_config REC_WITHOUT_CLOUD) == "yes" ]] ; then
            cd /home/app
            ./mp4record &
        fi
	sleep 4
        ./cloud &
    )
fi
```

alienatedsec, 2024-02-09:
There are many things to fix in the next pre-release. 

PR is the quickest way, but the next pre-release will take me longer than expected.

alienatedsec, 2024-02-10:
[New release - 0.4.1f](https://github.com/alienatedsec/yi-hack-v5/releases/tag/0.4.1f)

Minims, 2024-02-10:
So i have tried on Yi Home 27US 0.4.1f and I have still no watermark :-(.
Testing on Outdoor.

Minims, 2024-02-10:
The issue is still present on Yi OutDoor 9CUS. Cloud disable or not.

alienatedsec, 2024-02-10:
I'll test it later too

Minims, 2024-02-10:
I'm suspecting an issue with the `sleep` values and/or a process that is still alive and make the library load fails. But i can't confirm this.

alienatedsec, 2024-02-10:
@roleoroleo any advice on how the OSD should be called to be rendered?

roleoroleo, 2024-02-10:
It's probably a race condition.
If I remember correctly mmap.info should be changed between dispatch and rmm.
But not too early.
You should try to increase the sleep before to run rmm.

Minims, 2024-02-11:
While trying to debug, i have this. Don't know if it is related to this issue : 

```
/home/app #   LD_PRELOAD=/home/yi-hack-v5/lib/ipc_multiplex.so ./dispatch &
/home/app # insmod: can't insert '/home/app/localko/ssp_ms41909.ko': No such file or directory
```

roleoroleo, 2024-02-11:
> insmod: can't insert '/home/app/localko/ssp_ms41909.ko': No such file or directory

I think it's not related to the issue.

alienatedsec, 2024-02-11:
> While trying to debug, i have this. Don't know if it is related to this issue : 
> 
> 
> 
> ```
> 
> /home/app #   LD_PRELOAD=/home/yi-hack-v5/lib/ipc_multiplex.so ./dispatch &
> 
> /home/app # insmod: can't insert '/home/app/localko/ssp_ms41909.ko': No such file or directory
> 
> ```

Dispatch is used to load kernel drivers dynamically and its likely to have different or even similar errors on other cameras.

Minims, 2024-02-11:
@roleoroleo which one do you change ? the one at 3s before `rmm`?
In `/home/yi-hack-v5/script/system.sh` or in `/tmp/sd/yi-hack-v5/script/system.sh` ?

thanks.

alienatedsec, 2024-02-11:
> @roleoroleo which one do you change ? the one at 3s before `rmm`? In `/home/yi-hack-v5/script/system.sh` or in `/tmp/sd/yi-hack-v5/script/system.sh` ?
> 
> thanks.

`/tmp/sd/....`

Minims, 2024-02-11:
Hum, i have tried with 15s, 30s and 60s and i have stil no watermark on my cameras :-/
Quite strange. it seems working sometimes if copy/paste the block code manually once boot.
for info, killing `rmm` make my camera to reboot.

roleoroleo, 2024-02-11:
> for info, killing `rmm` make my camera to reboot.

This is normal.

Minims, 2024-02-12:
@roleoroleo Cloud disabled.

Minims, 2024-02-12:
@roleoroleo i have just tried with the cloud and the default sleep value, no change

roleoroleo, 2024-02-13:
Default sleep is probably not enough.
Try with 10 seconds, both with and without cloud.

Minims, 2024-02-13:
I can't find a working sleep :-/. 
Tried up to 120s ! with and without cloud. with and without audio stream.

alienatedsec, 2024-02-14:
> If it was working with 0.4.1d, we need to check what is changed...

@roleoroleo so, there were three major changes:
- wpa_supplicant upgraded to v2.10 - network related - still finding it odd that an image with this new version doesn't work on your `27US` and works on @Minims ' `27US` camera
- busybox upgraded to 1.36.1
- wsdd and onvif replaced with simple versions
- ptz and ipc changes

https://github.com/alienatedsec/yi-hack-v5/compare/0.4.1d...0.4.1h

Minims, 2024-02-14:
@roleoroleo @alienatedsec 

got it !!! 

[src/static/static/home/yi-hack-v5/etc/system.conf
](https://github.com/alienatedsec/yi-hack-v5/blob/d4466ff4cae1b3b3674a0bbc892266ee8a800caf/src/static/static/home/yi-hack-v5/etc/system.conf#L46)

TIME_OSD exists 2 times in the config file. So while parsing value is : 

```
yes
yes
```

and it does not enter the if.

Once i remove one TIME_OSD, it works while standard sleep.

Minims, 2024-02-14:
So i would say we can add 2 fix 

 - Remove one TIME_OSD from config file
 - Update `get_config()` with a `head -1` to avoid a similar issue
 
 ```
 get_config()
{
    key=$1
    grep -w $1 $YI_HACK_PREFIX/$CONF_FILE | cut -d "=" -f2 | head -1
}
```

Minims, 2024-02-14:
`head` is missing don't know how it can be replace.

alienatedsec, 2024-02-14:
> grep -w $1 $YI_HACK_PREFIX/$CONF_FILE | cut -d "=" -f2 | head -1

@Minims `head` is not present in this busybox compilation
https://github.com/alienatedsec/yi-hack-v5/blob/87620df69dfb82d24e433f6164a7ff741ff809af/src/busybox/.config#L263

Minims, 2024-02-14:
what about ? 

```
   grep -w $1 $YI_HACK_PREFIX/$CONF_FILE | cut -d "=" -f2 | awk 'NR==1 {print; exit}'
```

alienatedsec, 2024-02-14:
@Minims 

Adding HEAD is ~4 KB

[image]

Your command seems to be working ok
```
/home/yi-hack-v5 # grep -w TIME_OSD /tmp/sd/yi-hack-v5/etc/system.conf  | cut -d "=" -f2 | awk 'NR==1 {print; exit}'
yes
/home/yi-hack-v5 # grep -w TIME_OSD /tmp/sd/yi-hack-v5/etc/system.conf  | cut -d "=" -f2
yes
yes
/home/yi-hack-v5 #
```

Minims, 2024-02-14:
`awk` is powerful so it's OK keep this one. i think it's better to limit memory. 27US has already had issue with firmware size.

alienatedsec, 2024-02-14:
[image]

commit 06a7edfb (2024-02-14): [Pre-Release] 0.4.1i - Minor fixes #336 and versioning
 Changes to be committed:
	modified:   VERSION
	modified:   src/static/static/home/yi-hack-v5/etc/system.conf
	modified:   src/static/static/home/yi-hack-v5/script/check_conf.sh
	modified:   src/static/static/home/yi-hack-v5/script/check_update.sh
	modified:   src/static/static/home/yi-hack-v5/script/clean_records.sh
	modified:   src/static/static/home/yi-hack-v5/script/cloudAPI
	modified:   src/static/static/home/yi-hack-v5/script/cloudAPI_fake
	modified:   src/static/static/home/yi-hack-v5/script/conf2mqtt.sh
	modified:   src/static/stat [... cut]
closed 2024-02-14 by alienatedsec