alienatedsec/yi-hack-v5 #338: 27US stopped to ping, web server is not responding anymore

premiero, 2024-02-09:
About an hour after successfull flashing my camera with 0.3.8 firmware, it has stopped responding to ping. Web server is not responding anymore too. Cloud service is working without any issue, I can connect to camera via Yi app, so it's not bricked. I tried to reset camera - still no luck - no ping response, no web server. I toguht, maybe i should try reflash, but I have no idea how to force it?

 - Firmware Version: [0.3.8]
 - Device: [YI HOME 720P]
 - Model: [27US]

alienatedsec, 2024-02-10:
Looks like your camera has changed the IP. It's unlikely to be available online in the app and not on your network.

premiero, 2024-02-10:
Nope, it's not the case. I see when camera ask for IP from my DHCP server and receives it, also netstat on router shows active connections from this IP. Anyway - how to force camera reflash?

alienatedsec, 2024-02-10:
Please try the latest pre-release

premiero, 2024-02-10:
It's really strange - flashing successfull, but still no change (cloud service working, local access not)
The even starnger thing is that I can ping camera from my router, but not from any pc at the network. I successfully logged in to camera via ssh chain - ssh from pc to router, then from router ssh to camera. After login there is info:

` -----------------------------------------------------
  yi-hack-v5 - 0.4.1f
 -----------------------------------------------------
 WARNING: Cannot check for updates from GitHub.

 The camera might not be connected to the internet
 or, if it has just been turned on, the updater
 hasn't contacted the server yet.
 -----------------------------------------------------`

----- 15 minutes later -----

I tried ifconfig and iwconfig to see is there maybe something wrong there - but no, everything was as expected.
So I tought - maybe i will try to ping my PC from camera and ... it responded. Bump.
Than again I tried to ping camera from PC and guess what - it responded too.

So I have no idea what was wrong, but now seems like it's working again.

alienatedsec, 2024-02-10:
It feels like your router/network issue rather than a camera firmware problem.

premiero, 2024-02-10:
Maybe try to reproduce it as I described above.

alienatedsec, 2024-02-10:
> Maybe try to reproduce it as I described above.

Just tested to make sure I am not missing anything @premiero 

What is your router model and network config?

premiero, 2024-02-11:
Well, it's definitly something wrong with firmware, not the network. The problem came back after several hours, only now  I can't ping or log in even from a router and camera has a bootloop (keeps rebooting).

alienatedsec, 2024-02-11:
> only now I can't ping or log in even from a router and camera has a bootloop (keeps rebooting).

This I can confirm is the problem with the firmware. The other problem is still network-related. 

Which version did you run?

premiero, 2024-02-11:
As You suggested - 0.4.1f, but it was working ok for several hours at this version. Than - suddenly this happend and it was during normal usage, not during any changes. It simply started to reboot itself.

alienatedsec, 2024-02-11:
> As You suggested - 0.4.1f

That's useful, thanks. Please can you go with 0.4.1g as this version attempts to mitigate bootlooping issues?

premiero, 2024-02-11:
All right - flashed with 0.4.1g. It's working now without bootlop (we'll se for how long), but this strange bug with no reply stil remains. After more digging I found out, that camera stops to respond when I reboot camera from web server menu and after that I do not close the browser window. If I click reboot and than quickly close the window, than camera after reboot respond to ping and I can open new window and connect to web server without problem.

alienatedsec, 2024-02-11:
I am working on the next release and the potential fix for bootlooping - the `0.4.1g` is likely to fail too.

premiero, 2024-02-11:
I think also ONVIF serice is not working properly. I can play RTSP stream directly under VLC, but can't add camera at ONVIF related software (like SmartPSS or Synology Surveilance Station) - camera is not recognized as ONVIF capable, and manual  authentication fails.

[image]

alienatedsec, 2024-02-14:
The latest release `0.4.1i` please check to see if your issues are resolved anyhow - this model seems problematic for some

e.g., @roleoroleo has had wifi connection issues but @Minims worked fine

premiero, 2024-03-09:
I can confirm that 4.1i is stable at 27US, but final 4.1 relase is not. 4.1 hang after several minutes. 
Also - the bug with "unable to ping after reboot" is still present in every version.

alienatedsec, 2024-03-09:
@premiero 
Sorry. I can't replicate your issues, so if there isn't another user with the same symptoms you need to investigate your internal network configuration. I asked for details but it wasn't given; therefore, I am not sure how to help solve your case.

premiero, 2024-03-09:
There's noth special about network config. Just a static DHCP server (MAC lease based), WPA-2 wireless network IPv4 only, IPv6 disabled.
Maybe there is some problem because of loopback interface lack on camera? I cannot ping 127.0.0.1 nor it's wlan0 192.168.1.33 address from itself.
[image]
Rebooting router does not help, the only thing that help is what I descibed earlier - log via ssh from router to camera, send any packet (like ping) form camera to PC and magic, communication with camera is back.

alienatedsec, 2024-03-09:
What is the router?

premiero, 2024-03-09:
Asus RT-AX88U

alienatedsec, 2024-03-09:
Original firmware? OpenWRT?

alienatedsec, 2024-03-09:
Are you sure all your devices are defined on your DHCP server?

alienatedsec, 2024-03-09:
@premiero it sounds like the ARP table problem. I googled the problem which more or less confirms my initial response. Please check your network and I would start with your router, switches and DHCP servers.

premiero, 2024-03-10:
Thank You for trying to help me. It's original Asus firmware. There is only one router/switch/ap which is AX-88U. It also acts as DHCP server. ARP tables are correct (matches hardware camera MAC on both router and PC, I have checked it earlier). It is really stange, because for the first time, today I am unable to reproduce this error. I can acces webpage and ping responds event after reboot. But still, I cannot ssh to camera form my PC via putty (I can ssh to it from router). I started to suspect, that W10 is messing something around (maybe Defender issue?).
PS. Does Your camera respond to ping from itself?

alienatedsec, 2024-03-10:
> Does Your camera respond to ping from itself?

I checked earlier and it doesn't.

roleoroleo, 2024-03-21:
It's not a solution but we could add a
`ifconfig lo 127.0.0.1`
to fix the localhost interface.

closed 2024-04-28 by github-actions[bot]