alienatedsec/yi-hack-v5 #342: Stuck on "waiting to connect"

lbls55, 2024-02-11:
After putting new 0.4.1g hack on my Yi dome 1080p, everything was fine and after doing settings on GUI and reboot, the cam seems bricked.
I tried a reset with the rear button and now I'm stuck on "waiting to connect", using the app I do the connect process, I flash the QR code, the cam reboot and still stucked on "waiting to connect".
Any idea to unbrick?
Thanks for help.

alienatedsec, 2024-02-11:
Load earlier baseline. As you did a reset, you need to pair it with the YI app.

lbls55, 2024-02-11:
Ok, I will try  Envoyé à partir de Courrier pour Windows De : alienatedsecEnvoyé le :dimanche 11 février 2024 17:50À : alienatedsec/yi-hack-v5Cc : lbls55; AuthorObjet :Re: [alienatedsec/yi-hack-v5] Stuck on "waiting to connect" (Issue #342) Load earlier baseline. As you did a reset, you need to pair it with the YI app.

lbls55, 2024-02-11:
Before flashing ? I can’t connect, I’m stuck on « waiting to connect », I do the process, flash the QR code, everything is ok and the camera reboot and again « waiting to connect »… Envoyé à partir de Courrier pour Windows De : alienatedsecEnvoyé le :dimanche 11 février 2024 17:50À : alienatedsec/yi-hack-v5Cc : lbls55; AuthorObjet :Re: [alienatedsec/yi-hack-v5] Stuck on "waiting to connect" (Issue #342) Load earlier baseline. As you did a reset, you need to pair it with the YI app.

lbls55, 2024-02-11:
When I try to connect, hearing "waiting to connect", I do all the process, flash the QR code, hear "QRcode flashing is sucessfull", "connecting to wifi'', the blue button is flashing 10 seconds, then the yellow button flash 2 times and the camera reboot to "waiting to connect".

lbls55, 2024-02-11:
I've tried to reset two or three times...unsuccesfull.

alienatedsec, 2024-02-11:
Use the baseline from 0.3.8 and then try connecting. Don't reset as this removes the wifi config from the partition.

lbls55, 2024-02-11:
you mean only rootfs_h20 and home_h20 on the sd card? (0.3.8) No yi-hack-v5 folder?

lbls55, 2024-02-11:
uhhhhhh....done, paired again after flashing the 0.3.8 baseline files...on the road again, thanks for your help and for your hard work!!!

lbls55, 2024-02-11:
Still ready to put my fingers in the power outlet!!! ;-)

lbls55, 2024-02-11:
Working good now, back in your new baseline 0.4.1g, it's ok with Yi Dome 1080p 45US. Thanks again!!!

closed 2024-02-11 by lbls55