alienatedsec/yi-hack-v5 #343: ONVIF serice is not working properly

alienatedsec, 2024-02-11:
I think also ONVIF serice is not working properly. I can play RTSP stream directly under VLC, but can't add camera at ONVIF related software (like SmartPSS or Synology Surveilance Station) - camera is not recognized as ONVIF capable, and manual  authentication fails.

[image]

_Originally posted by @premiero in https://github.com/alienatedsec/yi-hack-v5/issues/338#issuecomment-1937860840_

alienatedsec, 2024-02-16:
> I found out, that the issue happend only when username/password is set at camera. If a user without password or user and password is set, than I cant use onvif. So to bypass this I have to leave empty both username and password.

@premiero Thanks for the additional info - is that still the case with `0.4.1i`?

Porkenstien, 2024-02-21:
Howdy,
in my testing, the ONVIF does not work as expected also.
The ONVIF apps are unable to find the device on my LAN. FYI, the app does see my other cameras(ONVIF compatible) and a couple of devices that are not ONVIF compatible. This leads me to the conclusion that the Yi Outdoor is not following the protocol properly. RTSP streaming does work from VLC Player, but only when there is _not a user and password_ in camera config
What can I provide to help dig into this deeper??

Yi Outdoor Cam(circa 2017)
Firmware: 3.0.0.0D
Hardware: 9CUS
yihack-v5: 0.4.1i
all testing from LAN

Minims, 2024-03-11:
I have tried to add my both camera to HA OnVif integration and ti fails too.
I have a user and password set.

FYI, my camera on Yi-hack Allwinner2 does not have the issue. Don't know if it share the same code for onvif.

closed 2024-04-18 by github-actions[bot]

mbox1, 2024-09-20:
Yi Outdoor 4CUS, yi-hack-v5 version 0.4.1 - same issue, ONVIF with password - not possible to add to Home Assistant, without user and password yes, but after several attempts - message - ONVIF server stops responding.