alienatedsec/yi-hack-v5 #350: [REQ] WiFi Recovery Guide

Hammerfest, 2024-02-18:
I could swear at one point there was a page for changing the WiFi info but I cannot seem to find it.

Can someone please post the instructions proper for yi-hack-v5 on how to change the WiFi info when a network change has occurred.

I tried to search through issues and discussions and found bits/pieces as well as conflicting information regarding directory structure placement and contents.

alienatedsec, 2024-02-18:
Main page of the camera at the very bottom, but this only applies to a camera with current WiFi connection. If you need to connect manually, please follow this https://github.com/alienatedsec/yi-hack-v5/issues/193

[image]

Hammerfest, 2024-02-18:
Yep, was trying to follow 193 but it did not work, tried putting the file in multiple places even.

alienatedsec, 2024-02-18:
The file needs to be in `/tmp/sd/recover/configure_wifi.cfg` as per changes in the commit https://github.com/alienatedsec/yi-hack-v5/commit/98f3db1ca38dcb3a81526412ded8103d8c2eb43d

Hammerfest, 2024-02-18:
Yep, I put it there as well

There was no folder called tmp so I made one and the subfolders
There was also a recover folder on root so I threw it in there as well
Also threw it in root as well

H:\configure_wifi.cfg
H:\tmp\sd\recover\configure_wifi.cfg
H:\recover\configure_wifi.cfg

The file only contained the 2 lines mentioned and tested with 2 different methods:
wifi_ssid=Bacon
wifi_psk=ExtraCrispy
or
wifi_ssid=[Bacon]
wifi_psk=[ExtraCrispy]

like I said, didnt work, placed it multiple places, I factory reset one of them and wiped it as well to put the latest 4.1x on it

alienatedsec, 2024-02-18:
Did you restart after placing the `configure_wifi.cfg` file inside the `/tmp/sd/recover` folder?

The recover folder should be in the root folder of your SD card. `/tmp/sd` is the location mounted on your camera.

Hammerfest, 2024-02-18:
The camera was powered down as it was not connectable due to wifi change, the sd card was removed and plugged into a PC

So the instructions I guess need to be to place the file in the recover folder located in the root of the drive, the instructions are unclear on source structure due to the whole linux mounting while connectable which is counter to changing wifi without access

Alas, it didn't work either and I have gone ahead and factory reset and installed latest

Instructions for anyone else who needs it:

WiFi Recovery Guide (when you cannot access it over the network)
On your computer use Notepad or something similar, create a file named:
configure_wifi.cfg

Inside this file paste the following lines:
wifi_ssid=[YourWiFiNetworkNameHere]
wifi_psk=[YourWiFiPasswordHere]

With camera powered down, remove MicroSD from camera and plug into your PC
Navigate with your file explorer to the MicroSD drive
On the root/top level of the MicroSD you should see a folder labeled "recover"
Copy the configure_wifi.cfg file into the recover folder
Move the MicroSD back to the camera and power on

For me it did not work, but I will text next time I have to move the camera's to a different wifi network.

alienatedsec, 2024-02-18:
Happy to accept any PRs and you contribute to the project :)

alienatedsec, 2024-02-18:
@Hammerfest This bothered me for the rest of the night and I think I found the problem. When the file is manually created, it adds `0D 0A` as hex values, a simple `ENTER`; however, UNIX will likely take `0D` as another empty character; hence, it's not working when applied when edited under Windows (or any other system) and applied under Linux. 

[configure_wifi.zip](https://github.com/alienatedsec/yi-hack-v5/files/14324957/configure_wifi.zip)

Please use and edit the attached, which ends each line with `0A` and try again.

This investigation [helped me fix another issue](https://github.com/alienatedsec/yi-hack-v5/commit/9226c4c19b8f02c6467af13f576c094c220581ac), so thanks :)

Hammerfest, 2024-02-19:
Sorry if bothered you enough for the evening! Glad either way it helped fix something else however!

Doesn't shock me, also with your example looks like I was correct the first time with removing the [ ] instead of just putting the text inside

Currently using them for a bit of work for a bit but in a few weeks I should be able to do more testing and force a need to network swap.

rikkistylz, 2024-03-25:
Hi so the camera blinked yellow and then turned blue but didn't connect back to the internet... And I tried resetting the cam and now I after scanning the QR code it won't connect to the internet again

fidesachates, 2024-04-04:
I've run into this now and need help. I tried to use this process and it didn't work. I then reflashed it with the original firmware from Yi and successfully reset and re-paired it using the Yi Home app. 

When I then go to flash it with this hack the 0.4.1 version by copying all the files into the root of my sd card, the camera flashes with the yellow light and then the blue light starts flashing indicating it's not connected to wifi. When I put the sd card back with the original firmware, it flashes and immediately connects to wifi without having to re-pair or reset. I can see the stream on the Yi Home app. No matter how often I flash the hack, it never connects, but the original firmware immediately connects. Any help please?

alienatedsec, 2024-04-04:
> P.S. when I try the configure_wifi.cfg, my sd card didn't have a recover folder so I manually created one. After reboot and the fail wifi connection, the file has disappeared from the recover folder.

@fidesachates 
The files disappear as it has served its purpose. The hack will try applying the setting on each boot if the file stays in the folder.

> Prior to all this, I was on 0.4.1 and it was connected just fine.

Are you sure the above is correct? did you mean a previous version to `0.4.1`?

fidesachates, 2024-04-04:
> @fidesachates
>
> The files disappear as it has served its purpose. The hack will try applying the setting on each boot if the file stays in the folder.

Yea I figured as much. Also I can hear the camera switch to night mode and on boot up it goes to my first bookmark position so I assume from all this that the camera is functioning; only the WiFi isn't working. 

> > Prior to all this, I was on 0.4.1 and it was connected just fine.
> 
> 
> 
> Are you sure the above is correct? did you mean a previous version to `0.4.1`?

Yes, I'm sure about this for two reasons. When is that I keep a detailed wiki for myself of my home network and my notes have 0.4.1 listed as the existing version. The second reason I'm certain that I was on 0.41 is this is the first version that I contributed to, so I remember upgrading. 

> Also, what are specifics of your WiFi network? WPA3, WPA2 or any mixed modes?

I have a Ubiquiti U7 Pro. The 2.4 GHz SSI ID is broken out from the others and is a mix of WPA2 and WPA3. I have two other YI Home hacks that are still connecting fine. Only the camera where I tried to change the Wi-Fi connection is broken. 

What do you make of the fact that on the original firmware the Wi-Fi connects but not on the hack firmware?

alienatedsec, 2024-04-04:
> What do you make of the fact that on the original firmware the Wi-Fi connects but not on the hack firmware?

@fidesachates do me a favour and extract the folder onto the `yi-hack-v5` folder of a WiFi-troublesome camera. Boot the camera and wait a few minutes until you either see the camera is connected or keep doing the same. It has to take a few minutes before you turn it off.

[wpa.zip](https://github.com/alienatedsec/yi-hack-v5/files/14871903/wpa.zip)

The hack will kick the following script
https://github.com/alienatedsec/yi-hack-v5/blob/d1997ae29ddeaf7d39d4d42c6da85901b7e56bcd/src/static/static/home/yi-hack-v5/script/system.sh#L31-L45

fidesachates, 2024-04-05:
> @fidesachates do me a favour and extract the folder onto the `yi-hack-v5` folder of a WiFi-troublesome camera. Boot the camera and wait a few minutes until you either see the camera is connected or [...]

It didn't seem to make a difference. I waited over 15 minutes and the blue light was still blinking. I did see that the wpa folder was gone and just the backups remained as per your script. 

Just a question, your script is moving *.so* but your zip folder didn't have any libs.

fidesachates, 2024-04-05:
Update:

After trying your suggestion with the wpa.zip, I retried [this](https://github.com/alienatedsec/yi-hack-v5/issues/350#issuecomment-1951490471) again to no avail. I then flashed myself back to 0.3.8 and this to my surprise brought the camera back online. I upgraded to 4.0 and things kept working. I tried to copy the 0.4.1 files to the sd card and that caused the camera again to not connect to wifi. 

At this point, I'd be tempted to tell you that you were right in suspecting I lied about the 0.4.1 working before I tried to switch my wifi network. However, I found in my wiki a note ( besides the version that is marked as 0.4.1) that says `(The default port is 80, but I uploaded previous configs that moved it to 8080)` which if I'm not mistaken was a 0.4.1 change.

However, if you want to proceed with debugging under the assumption that something is wrong with 0.4.1, I'd still be willing to help test out other debug steps you want. I feel like my camera broke due to one of these two reasons

1. [The wifi change procedure](https://github.com/alienatedsec/yi-hack-v5/issues/350#issuecomment-1951490471)
2. My new Ubiquity U7 Pro AP (again, the 2.5ghz is on a separate ssid). 

Both those two changes were made at the same time when the camera no longer connected.

alienatedsec, 2024-04-05:
> Just a question, your script is moving _.so_ but your zip folder didn't have any libs.

Libraries have not changed, so there is no need to upgrade.
@fidesachates We can try downgrading `wpa_supplicant` but I need the model number to extract the original files.

fidesachates, 2024-04-05:
> > Just a question, your script is moving _.so_ but your zip folder didn't have any libs.
> 
> Libraries have not changed, so there is no need to upgrade. @fidesachates We can try downgrading `wpa_supplicant` but I need the model number to extract the original files.

Is this what you're looking for? 65US

fidesachates, 2024-04-07:
Ah apologies. I have the Yi 1080P Dome

bzzlghtyr, 2024-04-15:
i'm interested in this as well. I have the Yi Home and am experiencing very similar issues where 3.8 works, but i get the loop of not connecting when I update to 4.1

alienatedsec, 2024-04-15:
> i'm interested in this as well. I have the Yi Home and am experiencing very similar issues where 3.8 works, but i get the loop of not connecting when I update to 4.1

https://github.com/alienatedsec/yi-hack-v5/issues/350#issuecomment-2037218326 have you tried this?

realealessio, 2024-04-30:
I have the same issue with the Yi Dome camera (v201).. Cannot connect to WIFI (WiFi password incorrect fake error)

jseabaugh, 2024-05-19:
Having the same issues with 0.4.1 with a 1080p home (48US). Tried both configure_wifi and wpa solutions in this thread (seperately and together) without success. Went back to 3.8 and it works.

closed 2024-06-26 by github-actions[bot]

jseabaugh, 2024-07-06:
Still only able to run 0.3.8. I get stuck in a boot loop (#390 ?) when trying to move to 0.4.1. Now I am running into the same issue as #312. Only way to resolve was to flash to an older version and then back to 0.3.8.

alienatedsec, 2024-07-07:
I plan to release a patch/reverse fix for those having problems with their WiFi connections after upgrading to `0.4.1`. I can't do it now as I am away for at least the next two weeks.

DayBit, 2024-08-07:
Hi, 

I have a 'Yi Dome' camera. For me, the latest version without the wifi problem is 0.4.1b. So, I think that there is a change in 0.4.1c that breaks wifi on some cameras.

I have switched to this software because the latest versions of the Yi app send ads as... motion detection notifications!!!!!. I was about to throw my three cameras in the trash and buy Tapo (TPLink). Now I am using this software in privacy mode :^) and a server that converts MQTT events to telegram messages. Thank you all very much!

alienatedsec, 2024-08-07:
> Hi,
>
> I have a 'Yi Dome' camera. For me, the latest version without the wifi problem is 0.4.1b. So, I think that there is a change in 0.4.1c that breaks wifi on some cameras.
>
> I have switched to this [...]

Thanks @DayBit 

I do realise it is about time to reverse the WPA upgrade I did, so people have options for their cameras to function properly.

rwijbenga, 2024-12-17:
Any update on the reverse of the WPA upgrade?