alienatedsec/yi-hack-v5 #366: Issues surrounding saving changes in Config pages (version 0.4.1)

Fookes74, 2024-02-29:
**Describe the bug**
Installed 0.4.1 as per instructions. On webpage I am unable to add a hostname (host name not valid) when saving, Wifi strength shows as 'no signal' even though camera is on the network and appears without issue in Yi Camera App.  Unable to modify any of the settings doesn't save. 

RTSP tab is completely blank. 

Baseline and firmware both 0.4.1 and Camera is Yi Outdoor 1080p 4CUS

**To Reproduce**
Steps to reproduce the behaviour:
N/A

**Expected behaviour**
Ability to input a new hostname, to make changes to settings and save and to enable RTSP feed.

**Screenshots**
Not required unless deemed necessary on response.

**Set Up Details (please complete the following information):**
 - Firmware Version: 0.4.1
 - Device: Yi Outdoor 1080p
 - Model: 4CUS

**Additional context**
None

**Technical Details**
None

alienatedsec, 2024-03-01:
That only indicates problems with your SD card. Try formatting the SD card using the YI app and copying the `yi-hack-v5` folder afterwards.

marteenas, 2024-03-14:
Having the same issue on Yi Dome 1080p. I can access the camera via SSH, I can modify system.conf using vi, it stores the values fine even after reboot (so SSD card is fine), but changing the values using WEB UI does not work. Using the same firmware as OP.

alienatedsec, 2024-03-14:
Can you execute `./tmp/sd/yi-hack-v5/script/launch.sh` and post the results?

marteenas, 2024-03-14:
Interesting thing is, when I connect via SSH I get:
`WARNING: Cannot check for updates from GitHub.

 The camera might not be connected to the internet
 or, if it has just been turned on, the updater
 hasn't contacted the server yet.`
 
 even though I can access the camera using Yi software via cloud :)

as for `/tmp/sd/yi-hack-v5/script/launch.sh`

> /home/yi-hack-v5 # /tmp/sd/yi-hack-v5/script/launch.sh
Using model: yi_dome_1080p
Version on cam: 0.4.1
Version on sd: 0.4.1
Specify the operation (k for kill, l for launch): l
Specify the settings (high, low, or both):

not sure what settings I want :)

alienatedsec, 2024-03-14:
how about `ls-l` command in `/tmp/sd/yi-hack-v5/etc`

```
/tmp/sd/yi-hack-v5/etc # ls -l
total 1792
-rwxr-xr-x    1 root     root           258 Feb 28 21:27 camera.conf
drwxr-xr-x    2 root     root        131072 Feb 28 12:58 crontabs
-rwxr-xr-x    1 root     root          2190 Feb 28 12:58 defaults.tar.bz2
drwxr-xr-x    2 root     root        131072 Feb 28 12:57 dropbear
-rwxr-xr-x    1 root     root             8 Feb 27 08:34 hostname
-rwxr-xr-x    1 root     root          3376 Feb 27 08:34 mqtt_advertise.conf
-rwxr-xr-x    1 root     root          1686 Feb 27 08:34 mqttv4.conf
drwxr-xr-x    2 root     root        131072 Feb 28 12:58 onvif_notify_server
-rwxr-xr-x    1 root     root            92 Feb 27 08:34 proxychains.conf
-rwxr-xr-x    1 root     root            92 Feb 28 12:58 proxychains.conf.template
-rwxr-xr-x    1 root     root            24 Feb 27 08:34 ptz_presets.conf
-rwxr-xr-x    1 root     root           704 Feb 27 08:34 system.conf
drwxr-xr-x    4 root     root        131072 Feb 28 12:58 wm_res
drwxr-xr-x    2 root     root        131072 Feb 28 12:58 wsd_simple_server
/tmp/sd/yi-hack-v5/etc #
```

marteenas, 2024-03-14:
> total 104
-rwxr-xr-x    1 root     root           257 Feb 24 21:31 camera.conf
drwxr-xr-x    2 root     root          8192 Feb 24 21:31 crontabs
-rwxr-xr-x    1 root     root          2190 Feb 24 21:31 defaults.tar.bz2
drwxr-xr-x    2 root     root          8192 Mar 13 15:23 dropbear
-rwxr-xr-x    1 root     root             0 Feb 24 21:31 hostname
-rwxr-xr-x    1 root     root          3363 Feb 24 21:31 mqtt_advertise.conf
-rwxr-xr-x    1 root     root          1657 Feb 24 21:31 mqttv4.conf
drwxr-xr-x    2 root     root          8192 Feb 24 21:31 onvif_notify_server
-rwxr-xr-x    1 root     root            92 Feb 24 21:31 proxychains.conf
-rwxr-xr-x    1 root     root            92 Feb 24 21:31 proxychains.conf.template
-rwxr-xr-x    1 root     root            24 Feb 24 21:31 ptz_presets.conf
-rwxr-xr-x    1 root     root           645 Feb 24 21:31 system.conf
drwxr-xr-x    4 root     root          8192 Feb 24 21:31 wm_res
drwxr-xr-x    2 root     root          8192 Feb 24 21:31 wsd_simple_server

marteenas, 2024-03-14:
if I edit system.conf in there /tmp/sd/yi-hack-v5/etc
using vi editor, then I get settings stored.

alienatedsec, 2024-03-14:
browser? clearing cache?

marteenas, 2024-03-15:
browser sends the request to camera without a problem:
POST:

> DISABLE_CLOUD: yes
REC_WITHOUT_CLOUD: no
RTSP: yes
ONVIF: no
ONVIF_WSDD: no
TIME_OSD: no
SNAPSHOT: yes
SNAPSHOT_LOW: no
SSHD: yes
FTPD: no
BUSYBOX_FTPD: no
TELNETD: no
NTPD: no
HTTPD: yes
PROXYCHAINSNG: no
SWAP_FILE: no
HOSTNAME: yidome1080phackv5

but the camera responds:
{error: "true"}

it still displays "Saved" in the web.

What I did is used Firefox instead and it saved the settings.
Then it started working on Chrome (Opera) browsers too. 
Probably some of the form inputs are not valid and the UI just does not display which ones (thus error: true).

closed 2024-04-22 by github-actions[bot]