alienatedsec/yi-hack-v5 #371: Cannot run 0.4.1 on 65US model

K13PTo, 2024-03-13:
**Describe the bug**
I barely got around to trying to update to version 0.4.1 but kept being unsuccessful.  It would sometimes show up on router RT-AC68U for split second. Reverted to 0.3.8

After installing 0.4.1, the camera does the following loop:
solid orange light
click sound
rotates camera as blue light flashes
returns to starting position for  camera
click sound
repeat

**Expected behaviour**
Boot into latest version

**Screenshots**
N/A

**Set Up Details (please complete the following information):**
 - Firmware Version: 0.4.1
 - Device: yi_dome_1080p
 - Model: 65US

**Additional context**
creates a file "log_no_wifi_dev.tar.gz" on SD card with logs. Sharing a modified version with network info removed in case it helps.
[log_no_wifi_dev.tar.gz](https://github.com/alienatedsec/yi-hack-v5/files/14584413/log_no_wifi_dev.tar.gz)

**Technical Details**

```
[./dispatch][1/1/0:0:18:5]: hw_type(2)
[./dispatch][1/1/0:0:20:625]: got sn(serialnumbergoeshere)[./dispatch][1/1/0:0:20:626]: got pwd(passwordgoeshere)[./dispatch][1/1/0:0:20:627]: got ssid(ssidgoeshere)[./dispatch][1/1/0:0:20:628]: got tnp_init_string(MMFBJPLJIDFHKMHBPLFLPDEEHACKFPGNHKEOKIIEIKBIDCBJPPINALNNPECOBLKLPNKJDKPBNNDABI)[./dispatch][1/1/0:0:20:735]: in choose_server, region_id = 17, api_server = https://api.us.xiaoyi.com, sname = familymonitor-h20, dlproto = mius
[./dispatch][1/1/0:0:20:736]: in choose_server, region_id = 17, api_server = https://api.us.xiaoyi.com, sname = familymonitor-h20, dlproto = mius
[./dispatch][1/1/0:0:20:756]: init ok, cost time(2770 ms)[./rmm][1/1/0:0:21:305]: f22 or f02
[./rmm][1/1/0:0:21:508]:  0xf
[./rmm][1/1/0:0:21:508]:  0x22
[./rmm][1/1/0:0:21:508]: enable adcr via himm 0x201200cc 0xFF033144[./rmm][1/1/0:0:21:626]: ai init ok!
[./rmm][1/1/0:0:21:627]: HI_MPI_AENC_AacInitHI_MPI_AENC_AacInit end
[./rmm][1/1/0:0:21:710]: rmm_init_aenc init ok!
[./rmm][1/1/0:0:21:720]: rmm_init_adec init ok!
[./rmm][1/1/0:0:21:721]: 
system call back libsns_f22.so_____________________
[./rmm][1/1/0:0:21:859]: sns_call_backsns_call_back
[./rmm][1/1/0:0:21:861]: HI_MPI_ISP_3A SUCCESS
[./dispatch][1/1/0:0:21:902]: wifi disconnected, now reconnect wifi
[./rmm][1/1/0:0:22:433]: isp_set_ae: HI_MPI_ISP_SetExposureAttr  target_lum =60
[./rmm][1/1/0:0:22:434]: 
_____Gamma test for table 4__________
[./rmm][1/1/0:0:22:436]: modify drc (150), time:22-446579[./rmm][1/1/0:0:22:444]: 
---------Rock_COMM_VENC_BindVpss VeChn = 0, VpssGrp = 0, VpssChn = 0
[./rmm][1/1/0:0:22:446]: rc set success! venc(0)
[./rmm][1/1/0:0:22:448]: 
---------Rock_COMM_VENC_BindVpss VeChn = 1, VpssGrp = 0, VpssChn = 1
[./rmm][1/1/0:0:22:450]: rc set success! venc(1)
[./rmm][1/1/0:0:22:453]: rmm_init success[./rmm][1/1/0:0:22:454]: font:/home/base/iso8859-1-24x48.bin
[./rmm][1/1/0:0:22:458]: filesize=36864
[./rmm][1/1/0:0:22:466]: init font /home/base/iso8859-1-24x48.bin ptr size(4)[./rmm][1/1/0:0:22:467]: font:/home/base/iso8859-1-8x16.bin
[./rmm][1/1/0:0:22:469]: filesize=4096
[./rmm][1/1/0:0:22:471]: init font /home/base/iso8859-1-8x16.bin ptr size(4)[./rmm][1/1/0:0:22:472]: load_font success[./rmm][1/1/0:0:22:475]: msg_proc success[./rmm][1/1/0:0:22:476]: zbar_proc success[./rmm][1/1/0:0:22:479]: pid[1471][./rmm][1/1/0:0:22:481]: pid[1504][./rmm][1/1/0:0:22:483]: pid[1506][./rmm][1/1/0:0:22:484]: pid[1505][./rmm][1/1/0:0:22:486]: pid[1510][./rmm][1/1/0:0:22:488]: pid[1509][./rmm][1/1/0:0:22:489]: AiGetLPCMStream start
 [./rmm][1/1/0:0:22:511]: pid[1508][./rmm][1/1/0:0:22:511]: rmm_vpp_create_bmp 5 success![./rmm][1/1/0:0:22:511]: VencGetH264Stream vechn=1
 [./rmm][1/1/0:0:22:512]: pid[1507][./rmm][1/1/0:0:22:512]: rmm_vpp_create_bmp 4 success![./rmm][1/1/0:0:22:513]: VencGetH264Stream vechn=0
 [./rmm][1/1/0:0:22:513]: pid[1511][./rmm][1/1/0:0:22:513]: modet_init img wid-hei:320-192, thresh:0
[./rmm][1/1/0:0:22:552]: set ai volume via himm 0x201200d8 0x0d0dc001[./rmm][1/1/0:0:22:664]: set ai volume via himm 0x201200c8 0x23c3c[./rmm][1/1/0:0:22:758]: set ao volume via himm 0x201200d4 0x03032424[./rmm][1/1/0:0:23:511]: ai_get_aacstream: get aenc stream select time out
[./rmm][1/1/0:0:24:520]: ai_get_aacstream: get aenc stream select time out
[./rmm][1/1/0:0:25:548]: ai_get_aacstream: get aenc stream select time out
[./dispatch][3/12/1:51:32:98]: DISPATCH_SET_DEFAULT_TIME 1710208292[./mp4record][3/12/1:51:35:246]: mp4 mode start init_finish[./cloud][3/12/1:51:36:364]: open share mem ok
[./mp4record][3/12/1:51:37:163]: got video main sps pps, ts=811310804[./mp4record][3/12/1:51:37:165]: got video sub sps pps, ts=811310804[./mp4record][3/12/1:51:37:165]: got audio aac config, ts=811311480[./mp4record][3/12/1:51:37:166]: mp4 mode got sd exist[./mp4record][3/12/1:51:37:166]: recordevent_init finish[./mp4record][3/12/1:51:37:167]: record_init finish[./mp4record][3/12/1:51:37:167]: got video main sps pps, ts=811310804[./mp4record][3/12/1:51:37:168]: got video sub sps pps, ts=811310804[./mp4record][3/12/1:51:37:168]: got audio aac config, ts=811311480[./p2p_tnp][3/12/1:51:37:169]: p2p_set_tnp_init_status 1 send_msg ok!
[./watch_process][3/12/1:51:37:825]: check_interval=10
[./watch_process][3/12/1:51:38:682]: process=dispatch;cmd=cd /home/app;./dispatch &;
[./watch_process][3/12/1:51:38:683]: process=cloud;cmd=cd /home/app;./cloud &;
[./watch_process][3/12/1:51:38:683]: process=rmm;cmd=reboot;
[./watch_process][3/12/1:51:38:684]: process=p2p_tnp;cmd=cd /home/app;./p2p_tnp &;
[./watch_process][3/12/1:51:38:684]: process=mp4record;cmd=cd /home/app;./mp4record &;
[./watch_process][3/12/1:51:38:685]: process=arp_test;cmd=cd /home/app;./arp_test &;
[./watch_process][3/12/1:51:38:685]: process=oss;cmd=cd /home/app;./oss &;
[./rmm][3/12/1:51:38:686]: modify drc (180), time:1710208298-632707[./rmm][3/12/1:51:38:922]: small action[0/3] (small:1)!  msarr[24][40]: 81, cnt:1/1, thresh:25, time:1710208298 sec
[./dispatch][3/12/1:51:39:838]: wpa may crashed, now reset
[./dispatch][3/12/1:51:40:170]: no usb dev
[./rmm][3/12/1:51:42:613]: isp_set_ae: HI_MPI_ISP_SetExposureAttr  target_lum =70
[./rmm][3/12/1:51:42:615]: isp_set_ae: HI_MPI_ISP_SetExposureAttr  target_lum =60
[./rmm][3/12/1:51:42:618]: modify drc (160), time:1710208302-632854[./rmm][3/12/1:51:42:623]: 
_____Gamma test for table 2__________
[./rmm][3/12/1:51:42:656]: modet_init img wid-hei:320-192, thresh:0

alienatedsec, 2024-03-13:
Which WiFi encryption have you enabled on the router?

K13PTo, 2024-03-13:
My default setting is AES on WPA2-Personal. I had tried WPA-Auto-Personal with TKIP+AES as well during troubleshooting to see if that would work. I also messed with preamble type short/long.  Went back to default after it failed to change anything.

scintillate, 2024-03-29:
Seeing the same issue on a 48US

alienatedsec, 2024-03-30:
@K13PTo @scintillate please can you share your full WiFi specs? That is to include the router model and its system e.g. factory or custom like OpenWRT. Also, do you use more than one router in your households? Anything additional like roaming is also relevant.

scintillate, 2024-03-31:
Router: TP-Link Archer C2600 v1.0
Router Firmware: 1.1.8 Build 20170306 Rel. 33259 (Factory)
Wireless Security: WPA/WPA2
Frequency: 2.4GHz

We use multiple routers, but not under the same SSID

K13PTo, 2024-04-01:
x2 ASUS RT-AC68U in AiMesh
Asuswrt-Merlin 386.12_6
Band 2.4 Ghz
Channel Bandwidth	20MHZ
Authentication	WPA2-Personal

alienatedsec, 2024-04-03:
> Router: TP-Link Archer C2600 v1.0 Router Firmware: 1.1.8 Build 20170306 Rel. 33259 (Factory) Wireless Security: WPA/WPA2 Frequency: 2.4GHz
> 
> We use multiple routers, but not under the same SSID

@scintillate have you considered upgrading to `OpenWRT`? https://openwrt.org/toh/tp-link/archer_c2600_v1
The factory firmware is old and could be full of bugs and vulnerabilities.

I got the same chipset, and a slightly different model `VR2600v1`, which I flashed for the same reason and it was running great since.
[image]

> I did at one point turn one off to see if it would affect connection

@K13PTo Its not about turning a one router off, but the mesh protocol to test if that is related.

alienatedsec, 2024-04-03:
@K13PTo I haven't enabled mesh networking protocol in the `wpa_supplicant` - the line is hashed in the current release.

https://github.com/alienatedsec/yi-hack-v5/blob/8b49002c447600bae81d949eba34ef554fae68d6/src/wpa/.config#L590-L591

I can build another version for you to test. Let me know if that is still required.

alienatedsec, 2024-04-04:
@K13PTo please try this https://github.com/alienatedsec/yi-hack-v5/issues/350#issuecomment-2037218326

K13PTo, 2024-04-08:
I disconnected 2nd router and did a hw reset followed by sw reset on 1st to do a full clear. So no more aimesh, still loops. It still shows up on wifi for brief second when its setting up new partitions on first run. I tried the wpa.zip file and left it running for undetermined amount, was busy. Still looped. If I add /recover/configure_wifi.cfg to the sd card a file gets created as "mtdblock2_19700101000014.bin".

closed 2024-05-15 by github-actions[bot]