alienatedsec/yi-hack-v5 #372: Issues to save information also Empty data - 6CUS

ABrauser, 2024-03-20:
i have flashed the 6CUS "yi Outdoor" camera. i have a connection, but everytime i set the hostname and some settings, after a reboot everything is gone. it looks like it cant access some files or data and cant save them.

Hostname | (none)
Hack (SD) Firmware Version | 0.4.1
Baseline Firmware Version | 0.4.1
Base Version | 3.0.0.0D_201809111054
Model Suffix | yi_outdoor
Hardware ID | 6CUS
Serial Number | XXXXXXXXXXXXX
PTZ | no
Local Time | Wed Mar 20 20:22:46 UTC 2024
Uptime | 0h 9m 12s
Load Average | 5.27 4.65 2.53
Free/Total Memory | 12856/30760 KB
Free SD Space | 100%

Does someone have a clue whats wrong with the camera?

alienatedsec, 2024-03-20:
How did you install the latest version? From scratch or by upgrading via Web from the earlier pre-release?

alienatedsec, 2024-03-20:
Which system did you use to extract files? Windows, MacOS, Linux?

ABrauser, 2024-03-20:
Windows 10 with 7-zip.

ABrauser, 2024-03-20:
i have also a yi_dome_1080p and in the folder /tmp/sd/ there is a yi_dome_1080p.conf created. but on the yi outdoor, where i have the problems no ".conf" file has been created inside /tmp/sd/ folder. i guess there must be something like yi_outdoor.conf, right?

alienatedsec, 2024-03-20:
> i have also a yi_dome_1080p and in the folder /tmp/sd/ there is a yi_dome_1080p.conf created. but on the yi outdoor, where i have the problems no ".conf" file has been created inside /tmp/sd/ folder. i guess there must be something like yi_outdoor.conf, right?

I have this

```
/tmp/sd/yi-hack-v5 # ls -l
total 1152
drwxr-xr-x    2 root     root        131072 Feb 28 12:58 bin
drwxr-xr-x    2 root     root        131072 Feb 28 12:58 busybox
drwxr-xr-x    7 root     root        131072 Mar 14 11:44 etc
-rwxr-xr-x    1 root     root             0 Feb 28 12:58 fw_upgrade_in_progress
drwxr-xr-x    2 root     root        131072 Feb 28 12:58 lib
drwxr-xr-x    2 root     root        131072 Feb 28 12:58 sbin
drwxr-xr-x    3 root     root        131072 Feb 28 12:58 script
drwxr-xr-x    4 root     root        131072 Feb 28 12:58 usr
-rwxr-xr-x    1 root     root             6 Feb 28 12:58 version
drwxr-xr-x    9 root     root        131072 Feb 28 12:57 www
/tmp/sd/yi-hack-v5 # cd ..
/tmp/sd # ls -l
total 66048
drwxr-xr-x    2 root     root        131072 Mar 20 19:12 log
drwxr-xr-x  171 root     root        131072 Mar 20 14:23 record
-rwxr-xr-x    1 root     root      67108864 Jan  1  1980 swapfile
-rwxr-xr-x    1 root     root            96 Feb 28 21:27 wd_rtsp.log
drwxr-xr-x   10 root     root        131072 Feb 28 12:58 yi-hack-v5
/tmp/sd #
```

No conf files

ABrauser, 2024-03-20:
but there should be a .conf file right? because there is one on my working yi_dome_1080p but not on the yi_outdoor.

alienatedsec, 2024-03-20:
The config is saved in the `/tmp/sd/yi-hack-v5/etc/system.conf`

ABrauser, 2024-03-20:
ah okay that makes sense. i think the yi dome 1080p was not fully "cleaned" i think thats a left over from some other hack. anyway. what i noticed after i have set a a hostname in the config (webview) the file "hostname" has still 0 kb. also all the settings from inside the system.conf are not loading in the gui. for example the TCP Ports are empty.  it looks like its not loading the values from the system.conf and saving changes back to it.
[image]

ABrauser, 2024-03-21:
any ideas why the webinterface is not readying the values from the system.conf and write them back?

ABrauser, 2024-03-21:
i manually changed the settings inside the system.conf. (changed http port, activated ftp, add timezone, activated rtsp and some other stuff) after a reboot now all the informations are loading correctly in the webinterface. looks like the problem is now solved.

alienatedsec, 2024-04-03:
This seems to be similar to #374

closed 2024-05-12 by github-actions[bot]