alienatedsec/yi-hack-v5 #374: No SSH, FTP and no config changes take effect after update to 4.1

waywit, 2024-03-23:
**Describe the bug**
I made a clean install on my both Yi Outdoor 1080p Cams (4DUS) with new SD Cards like described. Formatted the SDs in YI App and copied the folder back to sd card. Then I configured both cams from the scratch with hostnames, timezones and ... from the first start there was no ssd (activated on config), not rasp streaming (activated in config), no ftp (activated in config).

And I also mentioned that the settings made on the settings page does not be stored. Perhaps the whole thing has something to do with the other bugs about storing the config, but my config page looks like it is storing the values, but they do not have any effect. All ports stay closed on the camera.

**To Reproduce**
Steps to reproduce the behaviour:
1. Update Yi Outdoor 1080p 4CUS cam
2. Change values on config page and reboot
3. Config change are shown correctly on config page
4. But do not have any effect

**Expected behaviour**
Stored config changes take effect on the system, ssh access should work correctly.
Storing settings

**Screenshots**
The screenshot will show the expected state, but do not show the reality.

**Set Up Details (please complete the following information):**
Hack (SD) Firmware Version: 0.4.1
Baseline Firmware Version: 0.4.1
Base Version: 3.0.0.0D_201809111054
Model Suffix: yi_outdoor
Hardware ID: 4CUS

**Additional context**

**Technical Details**

alienatedsec, 2024-03-23:
It could be a duplicate #366 ; however, its slightly different as you see changes but those are not applying.

Can you post the content of your `/tmp/sd/yi-hack-v5/etc/system.conf` file? You can remove passwords and other sensitive data before posting

waywit, 2024-03-23:
Where can I get excel from ssh and telnet..? I can access the Cameras only with webui... the only open port is 80.
I am not sure if it makes sense to start over again...

I tried to reset one of my cams over the WebUI to yi-hack-defaults, which did not work and result in no WebUI. After formatting the sd card in Yi App, I can access the cam per FTP again and found a "yi-hack-v5" folder in /home folder... is this "normal"?

waywit, 2024-03-23:
Just one additional information... after resetting the cam I put the "yi-hack-v5" folder back to sd card and restarted the camera. Config is rested and ssh enabled but port is closed and no connect possible.

waywit, 2024-03-23:
:~$ nmap 192.168.1.241
Starting Nmap 7.80 ( https://nmap.org ) at 2024-03-23 12:59 CET
Nmap scan report for xyz (192.168.1.241)
Host is up (0.028s latency).
Not shown: 999 closed ports
PORT   STATE SERVICE
80/tcp open  http

Nmap done: 1 IP address (1 host up) scanned in 2.50 seconds

:~$ nmap 192.168.1.242
Starting Nmap 7.80 ( https://nmap.org ) at 2024-03-23 12:59 CET
Nmap scan report for yi-garten.fritz.box (192.168.1.242)
Host is up (0.032s latency).
Not shown: 999 closed ports
PORT   STATE SERVICE
80/tcp open  http

Nmap done: 1 IP address (1 host up) scanned in 1.46 seconds

alienatedsec, 2024-03-23:
Start the camera without SD and see if telnet is working - please send some screenshots of the file structure

waywit, 2024-03-23:
Hi, yes telnet and ftp are working without the sd card, but I am not sure why screenshots you need?

Here are some examples from ftp:

[image]

[image]

waywit, 2024-03-23:
Just tested a little bit around... and I can mount the sd card in my linux machine, so if you need some of the files I can get them from the card...
Here you can find the "system.conf" file, so all of my changes were stored there, but were not used to configure the system :-( 

`HTTPD=yes
TELNETD=no
SSHD=yes
FTPD=no
BUSYBOX_FTPD=no
DISABLE_CLOUD=no
REC_WITHOUT_CLOUD=no
MQTT=no
RTSP=yes
RTSP_STREAM=both
RTSP_AUDIO=no
ONVIF=no
ONVIF_WSDD=no
ONVIF_PROFILE=high
ONVIF_NETIF=wlan0
TIME_OSD=no
NTPD=yes
NTP_SERVER=fritz.box
PROXYCHAINSNG=no
SWAP_FILE=yes
RTSP_PORT=554
HTTPD_PORT=80
USERNAME=
PASSWORD=
TIMEZONE=CET-1CEST,M3.5.0,M10.5.0/3
FREE_SPACE=10
FTP_UPLOAD=no
FTP_HOST=
FTP_DIR=
FTP_DIR_TREE=no
FTP_USERNAME=
FTP_PASSWORD=
FTP_FILE_DELETE_AFTER_UPLOAD=no
CHECK_UPDATES=no
SSH_PASSWORD=
CRONTAB=
RTSP_ALT=no
SPEAKER_AUDIO=yes
SNAPSHOT=yes
SNAPSHOT_VIDEO=no
SNAPSHOT_LOW=yes
TIMELAPSE=no
TIMELAPSE_FTP=no
TIMELAPSE_DT=60
TIMELAPSE_VDT=
DEBUG_LOG=no`

I also found a "wd_rtsp.log" file in the root of the sd card containing many lines of:
`2024-03-23 12:57:36 - No running processes, restarting rRTSPServer ...`

alienatedsec, 2024-03-23:
What do you mean by saying you can mount the SD card? Dont you use Windows?

alienatedsec, 2024-03-23:
Before you answer the above, can you execute `vi` in the console using a telnet connection?

waywit, 2024-03-23:
Unfortunately no "vi" with telnet :-(

I have a linux server and there I tried to mount the sd card from my other yi cam... and it works :-) so I can access the full contents of the card... there I can see that my changes to the config are written to the files, but as stated above the system does not respect the config....

waywit, 2024-03-23:
I am not sure which was the previous installed version, but I think it was the 0.4.0 with the busybox issue preventing from OTA updates... before I also used several releases of the yi-hack without issues.

Should I try to downgrade the cameras?

waywit, 2024-03-23:
And the story continues ;-) I managed to update one of my Yi Outdoor Cams, a 4CUS model... I thought both were the same, but they are not. The other cam is a 9CUS and still not working, but now I have a different issue with the cam, it doesn't show "Baseline Firmware Version" in Web UI and the fat32error.txt is present on sd card. But the cam is accessible over ssh.

I formatted both cards with Windows 10 and FAT32 with standard cluster size and copied the same files to both sd cards. Is it possible that this has something to do with the sd card? I will try a different one tomorrow... Thank you very much for your help so far.

waywit, 2024-03-24:
Today I got the second cam working again. I have used the "old" sd card, freshly formatted and with the 3.8 files on it. After that it came back online with ssh and Yi App saying sd card is okay. Then I used the new sd card again, freshly formatted and equipped with actual 4.1 files on it... and after this it is back online again including ssh. Storing conf and settings is also possible... I have no idea what I made different to the first update. Except perhaps I gave the cams more time to boot and before changeing any setting?!?! But this is only an idea...

If you need further information from me, do not hesitate to ask... but for me personally I will accept this as an anomaly of the universe ;-)

jkworth, 2024-03-27:
I am currently having the same issue on my yi indoor 1080p U20 cameras. I do have to start with 0.3.8 but can only upgrade to 0.4.0. If i put 0.4.1 on, i can not alter the config at all and therefore not use it. For now I am staying with the 0.4.0. Let me know if there is any data/info you want to help troubleshoot. I can always put 0.4.1 back on a camera to collect any info you might need.

alienatedsec, 2024-03-28:
Can you check the permissions for `system.conf`?

jkworth, 2024-03-29:
i found 2 files with that name. here are the permissions on both:
```
/home/yi-hack-v5 # ls -al /tmp/sd/yi-hack-v5/etc/system.conf
-rwxr-xr-x    1 root     root           645 Feb 24 15:30 /tmp/sd/yi-hack-v5/etc/system.conf
/home/yi-hack-v5 # ls -al /home/yi-hack-v5/etc/system.conf
-rw-r--r--    1 root     root           645 Feb 24 18:49 /home/yi-hack-v5/etc/system.conf
/home/yi-hack-v5 # 
```

alienatedsec, 2024-03-30:
thanks @jkworth

I can't test the following but can you execute the below command and see if this fixes the problem?
```
chmod a+w /tmp/sd/yi-hack-v5/etc/system.conf
```

jkworth, 2024-03-30:
@alienatedsec I am unable to make the change. There is no error but changes do not seem to take affect. I assume this might be caused by whatever umask the mount is using when mounting the sd card.

```
/tmp/sd/yi-hack-v5/etc # ls -al sys*
-rwxr-xr-x    1 root     root           645 Mar 30 12:09 system.conf
/tmp/sd/yi-hack-v5/etc # chmod a+w system.conf 
/tmp/sd/yi-hack-v5/etc # ls -al sys*
-rwxr-xr-x    1 root     root           645 Mar 30 12:09 system.conf
```

alienatedsec, 2024-03-30:
@jkworth  I don't have any problems with two of my cameras. I will need to test the above and reproduce the problem myself. Otherwise, we are less-likely to fix the issue

jkworth, 2024-03-30:
@alienatedsec that also does not seem to work either. I am going to see if these work if I use 0.4.0 also.

jkworth, 2024-03-30:
@alienatedsec After some further pocking around here is what i found.

The permissions on the files when running 0.4.0 were the same as 0.4.1. I also could not change them but I could change the setting using the web page.
When running 0.4.1 I can edit the .conf files manually and restart and they take affect. However the web page does not read all the values nor can it change the settings. So I don't think file permissions are the issue.
I did notice that a `rmm` process is hogging the cpu for some reason when running 0.4.1.
Hope something in all this helps.

[image]

alienatedsec, 2024-03-30:
@jkworth how did you unpack the compressed `0.4.1` release? 
1. Using Windows and selecting `Extract all` which means only one step to get files. 
2. Using other extract tools e.g., 7zip - two step - extracting each compressed file
3. Linux - double-click on the package that extracts automatically

jkworth, 2024-03-30:
I use a Mac. So I just double click to unpack it.

alienatedsec, 2024-03-30:
> I use a Mac. So I just double click to unpack it.

@jkworth can you try other, more common operating system 😉

alienatedsec, 2024-03-31:
@jkworth 
I have redone two cameras in every possible way and couldn't reproduce the issue. Unfortunately, I don't have a Mac system to follow, and I suggest checking the following folders as well.

**`ls -l /tmp/sd/yi-hack-v5/script/`**

[image]

**`ls -l /tmp/sd/yi-hack-v5/www/cgi-bin/`**

[image]

jkworth, 2024-03-31:
@alienatedsec I used a different linux system to extract the files. I get the same result. I also went looking for logs but found none anywhere on the device. I even turned on the debug flag in system.conf in hopes for a log file somewhere.
Are there any logs to check?

[image]
[image]

alienatedsec, 2024-03-31:
@jkworth Notice your files modified date is different than mine even I have redone the process yesterday. There is something not right in the way the files are being extracted.

alienatedsec, 2024-03-31:
@jkworth Lets try a different approach

Do you have any output if you follow those two commands? The second one needs executing in that folder

```
cd /tmp/sd/yi-hack-v5/www/cgi-bin 
QUERY_STRING='conf=system' ./get_configs.sh
```

jkworth, 2024-03-31:
Here are the results of running the command. I have also tried a number of different ways to unpack the files on to the SD card. I can even see the dates similar to yours before i remove it from my computer. Once its in the camera they have different dates.

```
/home/yi-hack-v5 # cd /tmp/sd/yi-hack-v5/www/cgi-bin 
onf=system' ./get_configs.sh/tmp/sd/yi-hack-v5/www/cgi-bin # QUERY_STRING='conf=system' ./get_configs.sh
Content-type: application/json

{
"HTTPD":"yes",
"TELNETD":"no",
"SSHD":"yes",
"FTPD":"no",
"BUSYBOX_FTPD":"no",
"DISABLE_CLOUD":"no",
"REC_WITHOUT_CLOUD":"no",
"MQTT":"no",
"RTSP":"no",
"RTSP_STREAM":"high",
"RTSP_AUDIO":"no",
"ONVIF":"no",
"ONVIF_WSDD":"no",
"ONVIF_PROFILE":"high",
"ONVIF_NETIF":"wlan0",
"TIME_OSD":"no",
"NTPD":"no",
"NTP_SERVER":"pool.ntp.org",
"PROXYCHAINSNG":"no",
"SWAP_FILE":"no",
"RTSP_PORT":"554",
"HTTPD_PORT":"80",
"USERNAME":"",
"PASSWORD":"",
"TIMEZONE":"",
"FREE_SPACE":"10",
"FTP_UPLOAD":"no",
"FTP_HOST":"",
"FTP_DIR":"",
"FTP_DIR_TREE":"no",
"FTP_USERNAME":"",
"FTP_PASSWORD":"",
"FTP_FILE_DELETE_AFTER_UPLOAD":"no",
"CHECK_UPDATES":"no",
"SSH_PASSWORD":"",
"CRONTAB":"",
"RTSP_ALT":"no",
"SPEAKER_AUDIO":"yes",
"SNAPSHOT":"yes",
"SNAPSHOT_VIDEO":"no",
"SNAPSHOT_LOW":"no",
"TIMELAPSE":"no",
"TIMELAPSE_FTP":"no",
"TIMELAPSE_DT":"60",
"TIMELAPSE_VDT":"",
"DEBUG_LOG":"no",
"HOSTNAME":"",
"NULL":"NULL"
}/tmp/sd/yi-hack-v5/www/cgi-bin # 
````

I will try later today to dust off my windows machine and try using windows too,

alienatedsec, 2024-03-31:
Just to confirm, which browser are you using? @jkworth
Have you tried other browsers?

jkworth, 2024-04-03:
@alienatedsec Sorry for the delay but yesterday I finally got around to testing with windows. So i set my mac up to dual boot to windows 10. Used tar on the command like to unpack the tarball.  Everything worked just fine. 
This blows my mind that it works but it did. I set up 2 of my cameras with no issue. I have been a software engineer for over 20 years and never seen something like this. So I plan to investigate more in to what my mac is doing to cause it all to not work on the device. 

Thanks for the help.

alienatedsec, 2024-04-03:
I am glad it is working - closing this ticket for now.

> So I plan to investigate more in to what my mac is doing to cause it all to not work on the device.

Please, let me know how this ends

closed 2024-04-03 by alienatedsec

jkworth, 2024-04-03:
@alienatedsec I went back to the mac and re-downloaded the tarball using FF and chrome. While the OS showed differences in them. Both seemed to now work using my mac. So at this point i am going to assume there was something wrong with the first download or the unpacking the first time around.

While both worked here is the different tarballs. the one ending in -2 was downloaded by chrome and the other by FF.
[image]

Either way it all works. Even in my Mac. At this point I am going to blame high energy charged particles flipping bits on me. 🤣 👽