alienatedsec/yi-hack-v5 #379: recording to NVR disconnects

Zippeyrude, 2024-04-21:
After a few days the camera disconnected from hik nvr

The YIcamera app still shows the camera live vieW

A reboot resets and it connects and records as expected.  4.1j appears to be much more stable.

 - Firmware Version: [e.g. 4.1J
 - Device: [e.g., YI DOME 1080P]
 - Model: [e.g., 45US]

It's recording with onvif, which required rtsp enabled.

Can someone help point me to how to collect log files?

alienatedsec, 2024-04-21:
Please install the latest release.

Zippeyrude, 2024-04-22:
Hack (SD) Firmware Version | 0.4.1j
Baseline Firmware Version | 0.4.1j
Base Version | 1.9.2.0I_201812141405
Model Suffix | yi_dome_1080p
Hardware ID | 45US

As I am on the pre-release, the 'update to 0.4.1 states 'wrong baseline version'

[image]

alienatedsec, 2024-04-22:
Correct. You need to install it manually.