alienatedsec/yi-hack-v5 #382: 22EU Yi home 720p support

WojciechBednarski, 2024-05-12:
Hey,
Is there any chance for getting the support for Yi home camera 720p 22EU version?

alienatedsec, 2024-05-14:
Have you compared your firmware on the camera with the info in a table?

WojciechBednarski, 2024-05-14:
It seems to me that the one that is mentioned may be incorrect for the model I have. I tried the oldest firmware I found on the internet.
It seems like my camera is only accepting firmware files which are named home and rootfs, files named home_y18, rootfs_y18 are ignored by the camera. When I renamed them to home and rootfs, flashing was in progress, but camera refused to boot (putting back the original firmware restored it without problems).
Somewhere I found that this model is using Hi3518eV100. 
Firmware from https://github.com/npetrovski/yi-hack-easy worked kind-of, it enabled rtsp, but the camera was not stable and rebooted by itself, it's also not maintained anymore and there are no source files to compare.

brudnypiotr, 2024-06-07:
@WojciechBednarski would you be so kind as to share the original home and rootfs? I've bricked my camera trying to find a working solution for our 22EU version. My original firmware was `1.8.5.1M_201512011815`.

brudnypiotr, 2024-06-08:
For anybody looking for hope. I've unbriked my 22EU with these files. 
[22EU_Recovery.zip](https://github.com/user-attachments/files/15748673/22EU_Recovery.zip)

Stephonomon, 2026-08-30:
Second data point confirming @WojciechBednarski, on a **22US** unit (this thread is 22EU, so both regions of the same generation behave identically).

Camera: `YHS-113-IR`, Yi Home 720p, serial prefix `22US`, stock firmware `1.8.7.0F_201809191400`.

Copied `rootfs_y18` / `home_y18` / `yi-hack-v5` from 0.4.1 to a FAT32 card. The camera **ignored the firmware completely** — brief yellow, then a normal stock boot. Pulling the card afterwards showed it had been fully functional throughout: it recorded `record/*.mp4` (1280x720 h264 + aac) and `record_sub/*.mp4` (640x360) to that same card, wrote its `log_*.tar.gz` cloud dumps, and left `rootfs_y18` / `home_y18` byte-identical (sha256 verified before and after). So it mounted the card read-write and just never looked for those names — exactly as you described.

Useful detail if anyone wants to chase this properly: the stock `log_<id>_tnp_init_finish.tar.gz` on the card is a full system dump, and it confirms your Hi3518eV100 identification from the device side:

```
version=1.8.7.0F_201809191400   devtype=18E   provider=18E
Linux version 3.0.8 (gcc 4.4.1 (Hisilicon_v100...)) #1 Wed Apr 30 2014
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ)
Machine: hi3518    mem=40M   Chip:16MB SPI
mtdparts=hi_sfc:256k(boot)ro,128k(env),128k(conf),3072k(os),3584k(rootfs),9088k(home),128k(vd)
```

One thing worth correcting for anyone reading this later: the CPU is **not** the reason the images don't work. That was my first assumption too, given the project targets `Hi3518ev200`. But 0.4.1's binaries are built for this very core — decoding `.ARM.attributes` gives `Tag_CPU_arch=ARMv5TEJ` and `Tag_CPU_name=ARM926EJ-S` for `rRTSPServer`, `imggrabber`, `dropbear` and `libc.so.0`. Whatever stops the flash sits upstream of execution.

I did **not** try the rename, on the strength of your report plus @brudnypiotr's brick. Flagging that clearly for anyone who finds this thread: silently-ignored is the safe failure mode. Renaming to `rootfs` / `home` converts it into a camera that flashes and then won't boot.

Filed #473 with the full write-up, since the underlying documentation gap (firmware version alone matching the y18 baseline exactly, on hardware that isn't a y18) is what sends people here in the first place.