alienatedsec/yi-hack-v5 #385: No motion recording

johol, 2024-06-16:
**Describe the bug**
motion video is not recorded to sd card

**To Reproduce**
Looking at the event page no videos are shown 
and there are no recorded files on the cdcard

**Expected behaviour**
See videos

**Set Up Details (please complete the following information):**
 - Firmware Version: 3.0.0.0D_201809111054 / 0.4.1
 - Device:  yi_outdoor
 - Model: 6CUS

**Technical Details**
config:

/tmp/sd/yi-hack-v5/etc # cat system.conf
HTTPD=yes
TELNETD=no
SSHD=yes
FTPD=no
BUSYBOX_FTPD=no
DISABLE_CLOUD=yes
REC_WITHOUT_CLOUD=yes
MQTT=no
RTSP=yes
RTSP_STREAM=high
RTSP_AUDIO=yes
ONVIF=no
ONVIF_WSDD=no
ONVIF_PROFILE=high
ONVIF_NETIF=wlan0
TIME_OSD=yes
NTPD=yes
NTP_SERVER=pool.ntp.org
PROXYCHAINSNG=no
SWAP_FILE=no
RTSP_PORT=554
HTTPD_PORT=80
USERNAME=****
PASSWORD=****
TIMEZONE=CET-1CEST,M3.5.0,M10.5.0/3
FREE_SPACE=10
FTP_UPLOAD=no
FTP_HOST=
FTP_DIR=
FTP_DIR_TREE=no
FTP_USERNAME=
FTP_PASSWORD=
FTP_FILE_DELETE_AFTER_UPLOAD=no
CHECK_UPDATES=no
SSH_PASSWORD=****
CRONTAB=
RTSP_ALT=no
SPEAKER_AUDIO=no
SNAPSHOT=no
SNAPSHOT_VIDEO=no
SNAPSHOT_LOW=no
TIMELAPSE=no
TIMELAPSE_FTP=no
TIMELAPSE_DT=60
TIMELAPSE_VDT=
DEBUG_LOG=yes

johol, 2024-06-18:
Starting mp4record from the console I get following

MMAP address=0xb6d78000, size=1586752(0x183640), mode=0, try 1 
g_rcd_info.mmap_info->is_sd_exist =0

Seems to me, that mp4record does not recognize the SD card.
Could it be a issue with the  mmap.info file?

alienatedsec, 2024-07-07:
@johol are you sure you followed all flashing instructions?

Format the card in the yi app and copy the hack folder again.

johol, 2024-07-10:
Thanks for the advice. Formating the card with the app fixed recording. Noticed that the 128GB card is formated to exfat.

closed 2024-07-10 by johol