alienatedsec/yi-hack-v5 #386: Wifi Doesn't Connect on 0.4.1 - Yi Home 27US

tanmays, 2024-06-18:
**Describe the bug**
After flashing the latest 0.4.1 and trying to link using the Yi Life app, the camera scans the QR but is unable to connect and just restarts again with the same message "Waiting to connect". 

On flashing v0.3.8 - Everything works. So this issue is definitely limited to 0.4.1.

I'm using a brand new 32Gb FAT32 formatted SD card.

**Set Up Details (please complete the following information):**
 - Firmware Version: [0.4.1]
 - Device: [YI HOME]
 - Model: [27US]

**Technical Details**

```
[./dispatch][1/1/0:1:22:146]: hw_type(2)
[./dispatch][1/1/0:1:24:760]: open_ptz fail[./dispatch][1/1/0:1:24:779]: got sn(ClA98oXw3LWxALUwtln9AG6C4WOLCphE)[./dispatch][1/1/0:1:24:781]: got pwd()[./dispatch][1/1/0:1:24:782]: got ssid()[./dispatch][1/1/0:1:24:783]: got tnp_init_string()[./dispatch][1/1/0:1:24:784]: reset because no wifi config[./dispatch][1/1/0:1:24:787]: send_saveconfig_msg msg snd success[./dispatch][1/1/0:1:24:790]: get_config save conf
[./dispatch][1/1/0:1:24:868]: in choose_server, region_id = 17, api_server = https://api.us.xiaoyi.com, sname = familymonitor-y18, dlproto = mius
[./dispatch][1/1/0:1:24:868]: in choose_server, region_id = 8, api_server = https://api.xiaoyi.com.tw, sname = familymonitor-y18, dlproto = aws
[./dispatch][1/1/0:1:24:913]: init ok, cost time(2786 ms)[./dispatch][1/1/0:1:24:972]: got pwd()()[./dispatch][1/1/0:1:25:8]: got ssid()()[./dispatch][1/1/0:1:25:9]: config save finish 1[./dispatch][1/1/0:1:25:401]: save_config[./rmm][1/1/0:1:25:829]:  0x97
[./dispatch][1/19/6:41:36:98]: DISPATCH_SET_DEFAULT_TIME 1453185696[./rmm][1/19/6:41:36:180]: ai init ok!
[./rmm][1/19/6:41:36:183]: HI_MPI_AENC_AacInitHI_MPI_AENC_AacInit end
[./rmm][1/19/6:41:36:423]: rmm_init_aenc init ok!
[./rmm][1/19/6:41:36:427]: rmm_init_adec init ok!
[./rmm][1/19/6:41:36:435]: 
system call back libsns_ov9732.so_____________________
[./rmm][1/19/6:41:36:565]: sns_call_backsns_call_back
[./rmm][1/19/6:41:36:568]: HI_MPI_ISP_3A SUCCESS
[./rmm][1/19/6:41:36:743]: 
_____Gamma test for table 4__________
[./rmm][1/19/6:41:36:769]: 
---------Rock_COMM_VENC_BindVpss VeChn = 0, VpssGrp = 0, VpssChn = 0
[./rmm][1/19/6:41:36:773]: rc set success! venc(0)
[./rmm][1/19/6:41:36:776]: 
---------Rock_COMM_VENC_BindVpss VeChn = 1, VpssGrp = 0, VpssChn = 1
[./rmm][1/19/6:41:36:779]: rc set success! venc(1)
[./rmm][1/19/6:41:36:782]: rmm_init success[./rmm][1/19/6:41:36:784]: font:/home/base/iso8859-1-16x32.bin
[./rmm][1/19/6:41:36:786]: filesize=16384
[./rmm][1/19/6:41:36:790]: init font /home/base/iso8859-1-16x32.bin ptr size(4)[./rmm][1/19/6:41:36:794]: font:/home/base/iso8859-1-8x16.bin
[./rmm][1/19/6:41:36:795]: filesize=4096
[./rmm][1/19/6:41:36:797]: init font /home/base/iso8859-1-8x16.bin ptr size(4)[./rmm][1/19/6:41:36:800]: load_font success[./rmm][1/19/6:41:36:801]: msg_proc success[./rmm][1/19/6:41:36:803]: 
_____Gamma test for table 1__________
[./rmm][1/19/6:41:36:807]: pid[1511][./rmm][1/19/6:41:36:882]: set ai volume via himm 0x201200d8 0x1e1ec001[./rmm][1/19/6:41:36:932]: set ai volume via himm 0x201200c8 0x23c2e[./rmm][1/19/6:41:37:2]: set ao volume via himm 0x201200d4 0x03122424[./rmm][1/19/6:41:38:920]: HI_MPI_ADEC_SendStream ret 0xa0188040[./rmm][1/19/6:41:38:920]: msg snd success[./dispatch][1/19/6:41:38:922]: invalid msg 0x1002
[./rmm][1/19/6:41:39:21]: set ai volume via himm 0x201200d8 0x1e1ec001[./rmm][1/19/6:41:39:86]: set ai volume via himm 0x201200c8 0x23c2e[./rmm][1/19/6:41:39:148]: set ao volume via himm 0x201200d4 0x03122424[./rmm][1/19/6:41:40:954]: got RMM_SPEAK_WAIT[./cloud][1/19/6:41:43:238]: open share mem ok
[./p2p_tnp][1/19/6:41:43:849]: p2p_set_tnp_init_status 1 send_msg ok!
[./watch_process][1/19/6:41:44:589]: check_interval=10
[./watch_process][1/19/6:41:44:949]: process=dispatch;cmd=cd /home/app;./dispatch &;
[./watch_process][1/19/6:41:44:949]: process=cloud;cmd=cd /home/app;./cloud &;
[./watch_process][1/19/6:41:44:950]: process=rmm;cmd=reboot;
[./watch_process][1/19/6:41:44:950]: process=p2p_tnp;cmd=cd /home/app;./p2p_tnp &;
[./watch_process][1/19/6:41:44:950]: process=mp4record;cmd=cd /home/app;./mp4record &;
[./watch_process][1/19/6:41:44:951]: process=arp_test;cmd=cd /home/app;./arp_test &;
[./watch_process][1/19/6:41:44:951]: process=server;cmd=cd /home/app/web;./server &;
[./watch_process][1/19/6:41:44:951]: process=oss;cmd=cd /home/app;./oss &;
[./rmm][1/19/6:41:51:776]: decoded QR-Code symbol "b=SGo6qFskJAXo6JZU&s=VGFubWF5J3MgSW9U&p=AQF8cWFaWQl8bQ=="
[./rmm][1/19/6:41:51:778]: trans_json info=b=SGo6qFskJAXo6JZU&s=VGFubWF5J3MgSW9U&p=AQF8cWFaWQl8bQ==;key=&p=
[./rmm][1/19/6:41:51:779]: trans_json result=AQF8cWFaWQl8bQ==
[./rmm][1/19/6:41:51:780]: trans_json info=b=SGo6qFskJAXo6JZU&s=VGFubWF5J3MgSW9U&p=AQF8cWFaWQl8bQ==;key=&s=
[./rmm][1/19/6:41:51:781]: trans_json result=VGFubWF5J3MgSW9U
[./rmm][1/19/6:41:51:783]: trans_json info=b=SGo6qFskJAXo6JZU&s=VGFubWF5J3MgSW9U&p=AQF8cWFaWQl8bQ==;key=b=
[./rmm][1/19/6:41:51:784]: trans_json result=SGo6qFskJAXo6JZU
[./rmm][1/19/6:41:51:792]: rmm got /tmp/got_wpa ssid(Tanmay's IoT) pwd(9867206148) bind(SGo6qFskJAXo6JZU)
[./dispatch][1/19/6:41:51:809]: invalid msg 0x1005
[./rmm][1/19/6:41:51:830]: msg snd success[./dispatch][1/19/6:41:51:833]: in choose_server, region_id = 8, api_server = https://api.xiaoyi.com.tw, sname = familymonitor-y18, dlproto = aws
[./rmm][1/19/6:41:51:834]: msg snd success[./rmm][1/19/6:41:51:897]: set ai volume via himm 0x201200d8 0x1e1ec001[./dispatch][1/19/6:41:51:944]: msg snd success[./dispatch][1/19/6:41:51:944]: invalid msg 0x1004
[./rmm][1/19/6:41:51:954]: set ai volume via himm 0x201200c8 0x23c2e[./rmm][1/19/6:41:52:21]: set ao volume via himm 0x201200d4 0x03122424[./dispatch][1/19/6:41:52:338]: wifi disconnected, now reconnect wifi
[./rmm][1/19/6:41:54:79]: HI_MPI_ADEC_SendStream ret 0xa0188040[./rmm][1/19/6:41:54:80]: got RMM_SPEAK_SCAN_OK[./rmm][1/19/6:41:54:119]: set ai volume via himm 0x201200d8 0x1e1ec001[./rmm][1/19/6:41:54:155]: set ai volume via himm 0x201200c8 0x23c2e[./rmm][1/19/6:41:54:196]: set ao volume via himm 0x201200d4 0x03122424[./watch_process][1/19/6:41:54:607]: cloud crashed![./watch_process][1/19/6:41:54:643]: p2p_tnp crashed![./watch_process][1/19/6:41:54:662]: arp_test crashed![./watch_process][1/19/6:41:54:795]: server crashed![./cloud][1/19/6:41:54:809]: open share mem ok
[./p2p_tnp][1/19/6:41:55:433]: p2p_set_tnp_init_status 1 send_msg ok!
[./rmm][1/19/6:41:56:274]: HI_MPI_ADEC_SendStream ret 0xa0188040[./rmm][1/19/6:41:56:275]: got RMM_SPEAK_CONNECTTING[./dispatch][1/19/6:42:2:467]: wpa may crashed, now reset
[./dispatch][1/19/6:42:2:548]: no usb dev

```

bstrakt, 2024-07-08:
The same happens when upgrading Yi Dome Camera from 0.4.1b to latest 0.4.1 .. can't connect to wifi (neither after upgrading .. blue intermittent light .. neither after factory reset .. can't connect to Wifi). It worked before on all versions

ddunford, 2024-07-10:
I have the same problem on Yi Home 47US, 0.4.1b works fine, upgrading beyond 0.4.1c and a constant flashing blue light, factory resetting and attempting to reconnect results in wifi password wrong and never connects until downgrading again.

closed 2024-08-18 by github-actions[bot]