alienatedsec/yi-hack-v5 #396: Snapshots impossible to use

dreimer1986, 2024-08-30:
**Describe the bug**
Snapshots don't show up for quite a while on website and Home Assistant's FFMPEG runs into an timeout.

**To Reproduce**
Steps to reproduce the behaviour:
Check Snapshot site in webUI and grab some popcorn or do the same with camera.snapshot service under HA.

**Expected behaviour**
At least a bit faster, like the two internal Allwinner 2 based YI cams I use, too would be perfect.

**Screenshots**
-

**Set Up Details (please complete the following information):**
 - Firmware Version: 0.4.1
 - Device: Yi Outdoor
 - Model: 6CUS

**Additional context**
Add any other context about the problem here. If any other systems are involved, please complete the following:
 - Name of the system: Home Assistant
 - Version: 2024.9.0b1

**Technical Details**
Please add any technical details (e.g., log files)
I dont think you wanna see the ffmpeg error in HA and regarding a bunch of stale reports the problem is known already.

Is this chipset really that muich slower than the Allwiner 2? If not, maybe @roleoroleo has a clue why this whole thing is so overloaded and thus feels slow on website (for example on saving settings) and is way too slow to pass a snapshot to HA

alienatedsec, 2024-08-30:
@dreimer1986 Try using the YI method from the settings. At this moment, I will have no time to explore how to improve the speed, but I am happy to take any pull request under consideration.

dreimer1986, 2024-08-31:
I guess you mean the download way I did? Alright, already done. Not 100% stable as sometimes the file is 0 Bytes, but it works with some additional error collecting. I am no big Linux embedded guy, but I will go on a bug hunt, too. Maybe some luck and good karma will help me finding something :D

closed 2024-10-08 by github-actions[bot]