alienatedsec/yi-hack-v5 #397: Support for YI HOME 2K PRO

LexNetAT, 2024-09-03:
It is currently not possible to upload the firmware to camera [YI HOME 2K PRO](https://yitechnology.com/yi-home-2k-pro). It would be desirable if the new camera could also be operated via the hack.

I have the set of 4 and need a live stream without cloud and shorter latency.
I am also happy to provide technical support if someone give me instructions on what is needed or to do.

VisualBoy, 2024-09-21:
would be cool to have support for YI Home 2K.

For now I have installed this  hack: https://github.com/roleoroleo/yi-hack-Allwinner-v2     ([y623 version](https://github.com/roleoroleo/yi-hack-Allwinner-v2/releases/download/0.3.2/y623_0.3.2.tgz))

alienatedsec, 2024-09-22:
> would be cool to have support for YI Home 2K.
> 
> For now I have installed this hack: https://github.com/roleoroleo/yi-hack-Allwinner-v2 ([y623 version](https://github.com/roleoroleo/yi-hack-Allwinner-v2/releases/download/0.3.2/y623_0.3.2.tgz))

Nothing better than this version will work on your cameras.