alienatedsec/yi-hack-v5 #404: Where can I find the logs?

ad-on-is, 2024-10-15:
I'm investigating a bootloop issue, and need to see what's going on. Where can I find the logs. 

I have DEBUG_LOG=yes enabled

alienatedsec, 2024-10-15:
@ad-on-is The only way to investigate bootloop issue is to connect the camera via a serial terminal. That requires dismantling and soldering a couple of cables. 

First, try returning to a very early version and load the yi-hack-v5 `0.4.1` again.

ad-on-is, 2024-10-15:
@alienatedsec Yeah, I'm already aware of how to revert it back with an older root/home. It's happening almost every day, which Is why I want to debug it instead.

Are there any instructions for serial? I already have a serial2uart adapter for flashing esp8266, etc, if that's what is needed.

alienatedsec, 2024-10-15:
@ad-on-is How far did you go back? Which sources did you use?

ad-on-is, 2024-10-15:
@alienatedsec oooh... you mean an old version of yi-hack? no, I used an old stock firmware. and then put the latest 0.4.1 back on. I honestly never tried it with an older versionike 0.3.8, etc.

alienatedsec, 2024-10-15:
@ad-on-is Old stock firmware sounds good - in terms of serial connection have a look at this https://www.youtube.com/watch?app=desktop&v=uHColCH7SdA

ad-on-is, 2024-10-15:
@alienatedsec great. thank you. hope I find something.

ad-on-is, 2024-10-16:
@alienatedsec 
I was able to pull the logs. I hope this helps to identify the bootloop issue.

<details><summary>Logs</summary>
<p>

```
 done, booting the kernel.
Booting Linux on physical CPU 0
Linux version 3.4.35 (zhangbao@ANTS-SH-SV02) (gcc version 4.8.3 20131202 (prerelease) (Hisilicon_v300) ) #8 Fri Sep 29 11:23:43 CST 2017
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=00053177
CPU: VIVT data cache, VIVT instruction cache
Machine: hi3518ev200
Memory policy: ECC disabled, Data cache writeback
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 5588
Kernel command line: mem=22M console=ttyAMA0,115200 root=/dev/mtdblock4 rootfstype=jffs2 mtdparts=hi_sfc:256k(boot)ro,64k(env),64k(conf),1600k(os),1280k(rootfs),12992k(home),64k(vd1),64k(ver)
PID hash table entries: 128 (order: -3, 512 bytes)
Dentry cache hash table entries: 4096 (order: 2, 16384 bytes)
Inode-cache hash table entries: 2048 (order: 1, 8192 bytes)
Memory: 22MB = 22MB total
Memory: 18480k/18480k available, 4048k reserved, 0K highmem
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
    vmalloc : 0xc1800000 - 0xff000000   ( 984 MB)
    lowmem  : 0xc0000000 - 0xc1600000   (  22 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc0340000   (3296 kB)
      .init : 0xc0340000 - 0xc035cc64   ( 116 kB)
      .data : 0xc035e000 - 0xc037f220   ( 133 kB)
       .bss : 0xc037f244 - 0xc03b1850   ( 202 kB)
SLUB: Genslabs=13, HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
NR_IRQS:32
VIC @fe0d0000: id 0x00641190, vendor 0x41
sched_clock: 32 bits at 49MHz, resolution 20ns, wraps every 86767ms
Console: colour dummy device 80x30
Calibrating delay loop... 298.59 BogoMIPS (lpj=1492992)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 512
Initializing cgroup subsys freezer
CPU: Testing write buffer coherency: ok
Setting up static identity map for 0x8026ab08 - 0x8026ab60
dummy: 
NET: Registered protocol family 16
Serial: AMBA PL011 UART driver
uart:0: ttyAMA0 at MMIO 0x20080000 (irq = 5) is a PL011 rev2
console [ttyAMA0] enabled
uart:1: ttyAMA1 at MMIO 0x20090000 (irq = 30) is a PL011 rev2
uart:2: ttyAMA2 at MMIO 0x200a0000 (irq = 25) is a PL011 rev2
bio: create slab <bio-0> at 0
SCSI subsystem initialized
usbcore: registered new interface driver usbfs
usbcore: registered new interface driver hub
usbcore: registered new device driver usb
Switching to clocksource timer0
NET: Registered protocol family 2
IP route cache hash table entries: 1024 (order: 0, 4096 bytes)
TCP established hash table entries: 1024 (order: 1, 8192 bytes)
TCP bind hash table entries: 1024 (order: 0, 4096 bytes)
TCP: Hash tables configured (established 1024 bind 1024)
TCP: reno registered
UDP hash table entries: 256 (order: 0, 4096 bytes)
UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
NET: Registered protocol family 1
jffs2: version 2.2. (NAND) © 2001-2006 Red Hat, Inc.
fuse init (API version 7.18)
msgmni has been set to 36
io scheduler noop registered
io scheduler deadline registered (default)
io scheduler cfq registered
[... 1045 lines cut ...]
rtusb init rt2870 --->
create ok

usbcore: registered new interface driver rt2870
0x1300 = 00064300
Number of Packet Allocated in open = 1
Number of Packet Freed in open = 0
---extpkg.sh not exist---
---wpa_supplicant not exist---
rm: can't remove '/etc/resolv.conf': No such file or directory
ln: /etc/resolv.conf: No space left on device
mmz_start: 0x81600000, mmz_size: 10M
Hisilicon Media Memory Zone Manager
Module himedia: init ok
hi3518e_base: module license 'Proprietary' taints kernel.
Disabling lock debugging due to kernel taint
load sys.ko for Hi3518EV200...OK!
load tde.ko ...OK!
load region.ko ....OK!
ISP Mod init!

```

</p>
</details>

alienatedsec, 2024-10-16:
`ln: //.ash_history: No space left on device`

@ad-on-is here is the reason. Your rootfs partition is full. Did you move any files outside of `/tmp/sd/` folder?

Can you run `df -h` when it's not boot looping?

ad-on-is, 2024-10-16:
@alienatedsec 

I just figured, the issue was me (ofc).
When writing the ssh-tunnel script, I accidantely wrote to the root partition /logfile.txt... instead of /tmp/sd/yi-hack-v5/logfile.txt.

df -h shows  the root mount pont (/) at 90% now... 👍️

closed 2024-10-16 by ad-on-is