alienatedsec/yi-hack-v5 #406: 27US not load yi-hack-v5

gazambuja, 2024-10-28:
Im new here.
Yi Home 720 Camera 27US, last release (0.4.1), start without the hack, boot on default version without web access.

If I install yi-hack-v3 and v4 works fine, but I need RSTP.
Thanks.

alienatedsec, 2024-10-29:
@gazambuja I am sure you haven't followed the process - is your `yi-hack-v5` folder on the SD card?

gazambuja, 2024-10-30:
I fix the issue... is something about the mount of the sdcard on boot time:

```
# wget https://github.com/alienatedsec/yi-hack-v5/releases/download/0.4.1/yi_dome_0.4.1.tgz
# tar xvzf yi_home_0.4.1.tgz
# cd yi_home
# cp -r * /media/gustavo/8183-12AA 
# ls -lh /media/gustavo/8183-12AA                  
total 8.0M
-rw-r--r--  1 gustavo gustavo 7.1M Oct 30 09:10 home_y18
-rw-r--r--  1 gustavo gustavo 957K Oct 30 09:10 rootfs_y18
drwxr-xr-x 10 gustavo gustavo 8.0K Oct 30 09:11 yi-hack-v5
# umount /media/gustavo/8183-12AA

> Remove power to the camera, insert the microSD card, and turn the power back ON.
> The yellow light will come ON and flash for roughly 30 seconds, which means the firmware is being flashed successfully. The camera will boot up.

After this step, the blue led start blinking and stay blue.

# nmap 192.168.0.152
Starting Nmap 7.94SVN ( https://nmap.org ) at 2024-10-30 09:17 -03
Nmap scan report for 192.168.0.152
Host is up (0.52s latency).
Not shown: 998 closed tcp ports (conn-refused)
PORT   STATE SERVICE
21/tcp open  ftp
23/tcp open  telnet

Nmap done: 1 IP address (1 host up) scanned in 1.31 seconds

> Fantastic! I will conect using telnet:
# telnet 192.168.0.152
(none) login: root
Password: 
cat: can't open '/tmp/sd/yi-hack-v5/version': No such file or directory

      _     _           _           ___ 
  _ _|_|___| |_ ___ ___| |_ ___ _ _|  _|
 | | | |___|   | .'|  _| '_|___| | |__ |
 |_  |_|   |_|_|__,|___|_,_|    \_/|___|
 |___|
 -----------------------------------------------------
  yi-hack-v5 - 
 -----------------------------------------------------
 WARNING: Cannot check for updates from GitHub.

 The camera might not be connected to the internet
 or, if it has just been turned on, the updater
 hasn't contacted the server yet.
 -----------------------------------------------------

# /home/yi-hack-v5 # ls
etc      lib      script   version
# /home/yi-hack-v5 # ls /tmp/
.count               audio/               debug_alarm.txt      debug_p2p.txt        done                 log.txt              logsock              mmap.info            sd/                  wpa_supplicant.conf
MTK                  cloud_init_finish    debug_oss.txt        dns1                 gw1                  log_oss.txt          mask                 resolv.conf          view
# /home/yi-hack-v5 # ls /tmp/sd/
(empty)

> Ok... let me try to mount:
/home/yi-hack-v5 # mount /dev/mmcblk0 /tmp/sd
/home/yi-hack-v5 # ls /tmp/sd/
home_y18    rootfs_y18  yi-hack-v5

> Ok, let me add this to some script:
# vi /etc/init.d/S20yi-hack-v5
```

**> reboot and now is working!!!**

closed 2024-12-07 by github-actions[bot]

DarkwolfSGN, 2025-11-26:
> reboot and now is working

Hello, could you specific for a newbie how to do that ?