alienatedsec/yi-hack-v5 #418: SSID with special characters not working

danielhainich, 2025-02-07:
**Describe the bug**
Connecting to an ssid with special characters are not possible. F.e. ssid "SSI-D" isn't working

**To Reproduce**
Steps to reproduce the behaviour:
1. Try to connect to ssid with hyphen 
2. no connection can be established

**Expected behaviour**
ssid with special characters should work

**Set Up Details (please complete the following information):**
 - Firmware Version: [0.4.1j]
 - Device: [YI DOME 1080P]

alienatedsec, 2025-02-13:
I have special characters which works ok on the latest version. Don't use the `0.4.1j` version.

danielhainich, 2025-02-13:
Ok, but I upgrade to 0.4.1.j a while ago. How I can downgrade? Which version do you suggest,?

alienatedsec, 2025-02-14:
`0.4.1j` is a pre-release so the `0.4.1` will be the latest releases.

closed 2025-03-24 by github-actions[bot]