alienatedsec/yi-hack-v5 #420: Yi Outdoor[v_0.4.1] doesn't work

cigar, 2025-02-17:
Hardware: Yi Outdoor camera
Base Firmware: 3.0.0.0D_201809111054

I followed the guide and formatted the SD card with Windows FAT32. 
But it failed to reboot to blue light. and it restarts again and again.

Has anyone run into this problem?

rubenleon, 2025-03-03:
The same thing happened to me as it did to you, but in /tmp/sd/hackerror.txt, there was a link:
https://github.com/alienatedsec/yi-hack-v5/discussions/267 that solved all my problems.

I hope it helps you.

Best regards.

metcrl, 2025-04-01:
> Hardware: Yi Outdoor camera Base Firmware: 3.0.0.0D_201809111054
> 
> I followed the guide and formatted the SD card with Windows FAT32. But it failed to reboot to blue light. and it restarts again and again.
> 
> Has anyone run into this problem?

I have the same problem, did you fix it in any way?

DaibhidhG, 2025-04-15:
I'm having the same problem, but with Yi Home 47US camera

DaibhidhG, 2025-04-15:
cigar, [metcrl](https://github.com/metcrl) found the answer at https://github.com/alienatedsec/yi-hack-v5/issues/398.

cigar, 2025-04-16:
I'm delighted to hear this message from @DaibhidhG and can't wait to try it later. Thanks for sharing with me

closed 2025-05-24 by github-actions[bot]