alienatedsec/yi-hack-v5 #421: Yi 1080p Home - firmware not flashing

Nixellion, 2025-03-01:
Yi 1080p Home YYS 2016
Firmware: 2.1.0.0E_201809191630

Freshly bought. 
Followed instructions, but camera is not showing any signs of being flashed.  

Here's what I did:

- Formatted 16GB SD Card with Fat32
- Copied files from yi_home_1080p_0.4.1.tgz to the card
- Inserted into camera and powered it on
- Worth noting I did it on a fresh camera without connecting it to the app yet, does it matter?
- QR code, wifi connected
- Camera works as normal, but web page does not open, and there were no orange light flashing
- I tried restarting it a few times, and formatting SD card again and repeating the process - did not help

Result:

- Camera seems to be running stock firmware

Expected Result:

- Flashed yihack with web ui available

Nixellion, 2025-03-02:
[image]
[image]
[image]

Nixellion, 2025-03-02:
Looks like one of https://github.com/roleoroleo/yi-hack-MStar worked, I think it's y23, because I think yesterday I tried y203c and it did not work. As I was not sure I uploaded both today. It flashed and working now.

So it was rather confusing, because the number on the back say 6FUS which is mentioned on that page, but firmware version does not match the ones over there, instead it matches the one listed on this project "2.1.0.0E_201809191630"

alienatedsec, 2025-03-02:
> I guess this project should improve on it's list of supported cameras by adding numbers or a note about checking with those other projects to see if the number on the back matches one of those.

Happy to review your PR whenever you are ready.

closed 2025-03-02 by alienatedsec