alienatedsec/yi-hack-v5 #423: Anyone integrated Yi Dome 1080p in Home Assistant with motion detection?

bit111, 2025-03-08:
Hello,
I've integrated my Yi Dome 1080p in Home Assistant but I'm unable to deal with motion detection.
Has anyone managed to create an alert to send motion detection notifications with image?
Thanks.

JustLeader69420, 2025-03-08:
Hi, what have you tried? I'll quickly check, but I believe my motion detection appears when the camera is added through ONVIF, you could then just make a trigger to get a snapshot and send it to yourself.

JustLeader69420, 2025-03-08:
Yeah, mine detect motion almost instantly (binary_sensor.<cameraName>_motion_alarm) and clear in about 1 to 3 minutes.

bit111, 2025-03-08:
I've created this automation:

```yaml
triggers:
  - type: motion
    device_id: xxxxxxx
    entity_id: yyyyyyyy
    domain: binary_sensor
    trigger: device
actions:
  - action: camera.snapshot
    metadata: {}
    data:
      filename: /media/yi-cam{{ now().strftime("%Y%m%d-%H%M%S") }}.jpg
    target:
      entity_id: camera.yi_hack_v5_xxxxxxxcam
mode: single
```

The trigger is "a motion detection", but I wish to get the detected image in the action, instead of taking a new image.

bit111, 2025-03-08:
> I've created this automation:
> 
> triggers:
>   - type: motion
>     device_id: xxxxxxx
>     entity_id: yyyyyyyy
>     domain: binary_sensor
>     trigger: device
> actions:
>   - action: camera.snapshot
>     metadata: {}
>     data:
>       filename: /media/yi-cam{{ now().strftime("%Y%m%d-%H%M%S") }}.jpg
>     target:
>       entity_id: camera.yi_hack_v5_xxxxxxxcam
> mode: single
> 
> The trigger is "a motion detection", but I wish to get the detected image in the action, instead of taking a new image.

The previous action doesn't always work.
These are the errors I get:

```
[homeassistant.components.automation.motion_detected] Motion detected: Error executing script. Error for call_service at pos 4: Unable to get snapshot: Timed out after 10 seconds
home-assistant  | 2025-03-08 ERROR (MainThread) [homeassistant.components.automation.motion_detected] Error while executing automation automation.motion_detected: Unable to get snapshot: Timed out after 10 seconds
home-assistant  | 2025-03-08 ERROR (SyncWorker_8) [custom_components.yi_hack.camera] Fetch snapshot image failed from yi_hack_v5_xxxxx Cam, falling back to FFmpeg; HTTPConnectionPool(host='x.x.x.x, port=80): Read timed out.
```

JustLeader69420, 2025-03-08:
Maybe try bypassing the camera entity by getting the image using some kind of fetching thing to get it from the yicam's snapshot URL instead? I wouldn't know how to do it as I can barely organize my HA dashboards, but that could be a bit more reliable?

bit111, 2025-03-09:
This is exactly the problem: how can I fetch the snapshot without taking another one?

JustLeader69420, 2025-03-09:
I assume the camera is connected using ONVIF for the motion alerts? AFAIK, the motion alerts come from the camera as just that, an alert, not HA actively watching the camera for motion, no image is sent with an alert, so you'd have to get it yourself, hence the fetching.

bit111, 2025-03-09:
I know that the snapshot is accessible via `camera.yi_camera_name_motion_detection_image`. 
If I click on this "Motion Detection" sensor a popup with "Download Snapshot" button is displayed like the following:
[image]

Is there a way to get this saved image?

bit111, 2025-03-09:
I updated action of my automation in this way:
```yaml
action: notify.me
metadata: {}
data:
  message: Movement detected!
  data:
    photo:
      - file: /api/camera_proxy/camera.yi_camera_name_motion_detection_image
        caption: Movement detected!
```
but no message is sent and in HA logs I have these error:
```
home-assistant  | 2025-03-09 WARNING (MainThread) [homeassistant.components.telegram_bot] '/api/camera_proxy/camera.yi_camera_name_motion_detection_image' are not secure to load data from!
home-assistant  | 2025-03-09 ERROR (MainThread) [homeassistant.components.telegram_bot] Can't send file with kwargs: {'message': "Movement detected", 'file': '/api/camera_proxy/camera.yi_camera_name_motion_detection_image', 'caption': 'Movement detected'}
```

JustLeader69420, 2025-03-09:
> I know that the snapshot is accessible via `camera.yi_camera_name_motion_detection_image`. If I click on this "Motion Detection" sensor a popup with "Download Snapshot" button is displayed like the following: <img alt="Image" width="212" src="https://private-user-images.githubusercontent.com/6236258/420695149-a627aafe-4476-4eed-99d4-4d3f55cab326.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDE1NDc1OTQsIm5iZiI6MTc0MTU0NzI5NCwicGF0aCI6Ii82MjM2MjU4LzQyMDY5NTE0OS1hNjI3YWFmZS00NDc2LTRlZWQtOTlkNC00ZDNmNTVjYWIzMjYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDMwOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAzMDlUMTkwODE0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YzM5ZDA4YTFhNjY5MDI4YzY2OGFiY2IyNzE3MjA4MzYxODVhZmNlYTgzOTA0OTYyMWFlODgyMWVhMWU3MTQzMiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.UFlK2N-xeRUsfbShDKreXlCKA93V6BKLWttMU5x-RDM">
> 
> Is there a way to get this saved image?

I would assume that's just a button to fetch a snapshot automatically, would have to check, are you sure it doesn't just get a new snapshot every time you click it? The same button is in the camera stream view, AFAIK, it just gets a snapshot from the camera (it is a snapshot from its snapshot URL as it isn't the same quality as the stream (I force low quality snapshots))

Why would it matter that a snapshot is 1-2 seconds old? A person would still likely be in frame, if the snapshot is even that delayed. I still believe HA has nothing to do with the motion detection and just shows the status the camera gives it, independent of the camera stream. I don't see a reason HA should be doing the motion detection; yeah you can setup an action to take a snapshot the instant the status changes to motion detected, but I'd say that's about it.

JustLeader69420, 2025-03-12:
Then I wouldn't know, sorry

JustLeader69420, 2025-03-16:
Hello, while setting up a motion activated light using the motion detection of my camera, the ONVIF-provided motion detection seems to get stuck (or unknown) sometimes, so I thought to setup MQTT, hopefully it will prove to be more reliable, anyways, after the camera was added using MQTT as well, it has added an entity that seems to provide the snapshot of the last motion detection, camera.yi_camera_motion_detection_image, it always seems to be just the last motion detection image no matter how many times it is opened, it's not the fastest (5-10 seconds first open, under a second on repeated attempts, so perhaps it gets cached in the browser), but neither is my HA instance. I don't know how the entity operates, if it gets it from the camera or from HA storage, but it seems to load every time; you can try setting up MQTT, it's not too difficult, I can help if you need.

dreimer1986, 2025-04-15:
I always ran into the problem that the cam seems to be too slow to handle the snapshot fast enough and most of the time the image was just a 0 kb file. After some discussion in here I did some trickery to get images quite reliable:

```
shell_command:
  doorbellphotobig: wget http://192.168.181.116/cgi-bin/snapshot.sh\?res\=high --timeout=600 -O /config/downloads/doorbell.jpg
  doorbellphotosmall: wget http://192.168.181.116/cgi-bin/snapshot.sh\?res\=low\&watermark\=yes --timeout=600 -O /config/downloads/doorbell.jpg
```

```
- id: nightalarm
  alias: NachtAlarm
  triggers:
  - entity_id: binary_sensor.yi_hack_v5_f26987_motion_detection
    from: 'off'
    to: 'on'
    trigger: state
  conditions:
  - condition: state
    entity_id: sun.sun
    state: below_horizon
  actions:
  - action: shell_command.doorbellphotosmall
    data: {}
  - data:
      title: Home Assistant
      message: Photo
      data:
        photo:
          file: /config/downloads/doorbell.jpg
          caption: Eingang nachts
    action: notify.telegrambot
  mode: single
```

Still not the fastest solution, but at least now I get a image that is close to the moment where the motion detecton was triggered.

bit111, 2025-05-25:
> Hello, while setting up a motion activated light using the motion detection of my camera, the ONVIF-provided motion detection seems to get stuck (or unknown) sometimes, so I thought to setup MQTT, [...]

Hi, sorry for not answering earlier.
Could you give me an example of automation?

JustLeader69420, 2025-05-28:
> Hi, sorry for not answering earlier. Could you give me an example of automation?

Hi, I don't believe I had to setup an automation, I just installed an MQTT broker in HA and added it in the HA MQTT section of the camera's settings.