alienatedsec/yi-hack-v5 #424: New Yi camera missing UART points

lilkuz2005, 2025-03-22:
The wife ordered a couple of these Yi cams from TEMU a couple weeks ago. I wanted to use them with rtsv but quickly found out they are locked to the Yi app. I took one of the cams apart to see if I could locate the UART ports and I can't seem to find them on this model. Ill upload a couple photos to show the PCB. The MCU is a ANYKA ak3918en080 v200. The silkscreen is mostly labeled but I didn't see any TX RX points. Anyone have any ideas if this model supports UART ?

[image]

[image]

lilkuz2005, 2025-03-26:
Messing around with this camera again today. Probing around on the PCB I found the UART points. Its the 3 through hole points that are unpopulated right below the CPU, the pins are from left to right on the CPU side of the board RX TX GND. Baud rate is 115200

U-Boot 2013.10.0-AK_V2.0.04 (Jan 29 2024 - 11:07:05)

DRAM:  64 MiB
8 MiB
Create flash partition table init OK!
ANYKA SDHC/MMC4.0: 0
Load Env CRC OK!
In:    serial
Out:   serial
Err:   serial

Hit any key to stop autoboot:  0

SF: 1417152 bytes @ 0x31000 Read: OK
## Booting kernel from Legacy Image at 81808000 ...
   Image Name:   Linux-3.4.35
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    1417088 Bytes = 1.4 MiB
   Load Address: 81808000
   Entry Point:  81808040
   Verifying Checksum ... OK
   XIP Kernel Image ... OK

Starting kernel ...

Uncompressing Linux... done, booting the kernel.
Anyka Linux Kernel Version: 2.5.04
Booting Linux on physical CPU 0
Linux version 3.4.35 (ljy@ants-szfir) (gcc version 4.8.5 (anyka (gcc-4.8.5 + binutils-2.24 + ulcibc-0.9.33.2)(20170223)) ) #11 Thu Jun 20 16:18:24 CST 2024
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=00053177
CPU: VIVT data cache, VIVT instruction cache
Machine: Cloud39EV2_AK3918E80PIN_MNBD
Memory policy: ECC disabled, Data cache writeback
ANYKA CPU AK3918 (ID 0x20150200)
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 10160
Kernel command line: console=ttySAK0,115200n8 root=/dev/mtdblock4 rootfstype=squashfs init=/sbin/init mem=64M memsize=64M
PID hash table entries: 256 (order: -2, 1024 bytes)
Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
Memory: 40MB = 40MB total
Memory: 36436k/36436k available, 4524k reserved, 0K highmem
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
    vmalloc : 0xc3000000 - 0xff000000   ( 960 MB)
    lowmem  : 0xc0000000 - 0xc2800000   (  40 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc03a2000   (3688 kB)
      .init : 0xc03a2000 - 0xc03bc000   ( 104 kB)
      .data : 0xc03bc000 - 0xc03e0d18   ( 148 kB)
       .bss : 0xc03e0d3c - 0xc0401be8   ( 132 kB)
SLUB: Genslabs=13, HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
NR_IRQS:95
sched_clock: 32 bits at 12MHz, resolution 83ns, wraps every 357913ms
AK39 console driver initial
console [ttySAK0] enabled
Calibrating delay loop... 199.06 BogoMIPS (lpj=995328)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 512
CPU: Testing write buffer coherency: ok
Setting up static identity map for 0x81ad6528 - 0x81ad6580
devtmpfs: initialized
NET: Registered protocol family 16
On-chip L2 memory initialized
AK39 clocks: CPU 400MHz, MEM 200MHz, ASIC 100MHz
Anyka platform share gpio locks initialize.
bio: create slab <bio-0> at 0
usbcore: registered new interface driver usbfs
[... 135 lines cut ...]
Freeing init memory: 104K
mount all file system...
start telnet......
starting mdev...
**************************
    Love Linux ! ! !
**************************
200+0 records in
200+0 records out
102400 bytes (100.0KB) copied, 0.028442 seconds, 3.4MB/s
camera_setup:not factory mode
sensor_gc1084.ko
isp_gc1084.conf
aksensor_module_init
id 65535
aksensor 0-0001: Sensor ID error

anyka login: close white led, whiteLightNegativeFlag = 0
/usr/sbin/service.sh: line 235: can't create /sys/user-gpio/WHITE_LED: nonexistent directory
==================/etc/jffs2 used space more tharn 50
ifconfig: SIOCGIFFLAGS: No such device
cp: can't stat '/usr/sbin/hostapd_atmb.conf': No such file or directory
start ipc service......
start ipc service......
start net service......

closed 2025-05-03 by github-actions[bot]

jg-rmr, 2025-11-15:
Login: root
Password: yunyi666