alienatedsec/yi-hack-v5 #427: YI Outdoor - can't install v5

metcrl, 2025-03-31:
Hi guys,

I am trying to install V5 because I would need RTSP for two of my YI Outdoor.
The problem is that after I put the two files and folder in the sd, the camera reboots and blinks yellow (I assume fimware installation), but then it goes into infinite loop where: steady yellow light for a few seconds, blinking blue light, reboot, steady yellow light again, blinking blue light again, and so on.

I can't figure out where I'm going wrong. I have also tried installing V4 and V3 and they work without any problems.

DaibhidhG, 2025-04-15:
I have a YI Home, 47US, and I'm having the same issue.
Yi-Hack-4 worked fine.
Happy to do any testing required to assist in finding a fix.

Thanks

metcrl, 2025-04-15:
I solved following #398 

- Download the correct firmware for your camera
- Copy everything to the microsd (properly formatted)
- Download the wpa.zip file inside #398 
- Follow https://github.com/alienatedsec/yi-hack-v5/issues/398#issuecomment-2361545011 to copy the content of the zip into the microsd
- Reboot the camera

Remember that you **MUST** keep the _yi-hack-v5_ folder inside the microsd.
If you have problems with video recording, format the microsd from the YI Home app, then put the yi-hack-v5 folder back in the microsd (still with the wpa folder from the zip) WITHOUT the home and rootfs files.

DaibhidhG, 2025-04-15:
Brilliant, thanks metcrl, that's sorted it

closed 2025-05-24 by github-actions[bot]